uzh-rpg/rpg_svo #65: Intended live camera output?

clauniel, 2015-01-09:
I use the following sequence of commands to launch the live camera svo using ros.

    $ roscore
    $ roslaunch svo_ros live.launch
    $ rosrun rviz rviz
    $ rosrun rqt_svo rqt_svo

I can see svo starting and saying it finds an image source on /image_raw, but nothing happens in either rviz or rqt_svo. rqt_svo says "not connected", but I can press "start" which does nothing other than printing "START SLAM" to the terminal and "quit", which prints "QUIT SLAM".

My cpu runs at 100% however, so I assume something is happening, but I see no output.

taogashi, 2015-01-09:
it shows you have no image topic published. you can take the .bag file offered by cfo for test purpose, which suites the defualt camera parameters. `rosbag play xx.bag` and then check ros topics again, it should work. For live camera input, you need packages such as [usb_cam](http://wiki.ros.org/usb_cam)

clauniel, 2015-01-09:
I previously ran the bag file example and it worked fine, but it doesn't run anymore because the .bag file is somehow now a wrong version. I also have the usb_cam package.

MuiLe, 2015-04-05:
I have the same problem:
Topics are:

/camera/image_raw
/rosout
/rosout_agg
/svo/dense_input
/svo/image
/svo/info
/svo/keyframes
/svo/points
/svo/pose
/svo/remote_key
/tf

Could someone help us.
Thanks,

MuiLe, 2015-04-13:
I added an ROS node in live.launch and it seem to works now.

<node name="camera" pkg="usb_cam" type="usb_cam_node" output="screen" >
        <param name="video_device" value="/dev/video0" />
        <param name="image_width" value="640" />
        <param name="image_height" value="480" />
        <param name="pixel_format" value="yuyv" />
        <param name="framerate" value="30" />
        <param name="camera_frame_id" value="webcam" />
    </node>

But could someone let us know what the output message is in case we succeed to estimate the came pose.

Thanks.

sj312931, 2015-06-17:
@clauniel I am having the same problem. rqt_svo always shows not connected. Could you please tell me how you solved the problem? I see the following topics - 
/camera/image_raw
/clicked_point
/initialpose
/move_base_simple/goal
/rosout
/rosout_agg
/svo/dense_input
/svo/image
/svo/info
/svo/keyframes
/svo/keyframes_array
/svo/points
/svo/points_array
/svo/pose
/svo/remote_key
/tf
/tf_static

clauniel, 2015-06-25:
@sj312931 I did not find a solution, but the problem fixed itself when I switched from Arch Linux to Ubuntu.

ank700, 2015-11-16:
Hello, 
  I have the same problem as clauniel. I am able to launch live.launch correctly but on the rqt window, it just shows 'not connected'. I tried to change the SVO namespace from default to svo and still nothing happens.

What does 'not connected' signify?

Can anyone please help me with this problem? Any help is much appreciated.
Hi @cfo, maybe you can suggest something.

Thanks in advance.

astridpetitjean, 2017-03-02:
Hi, 

In my case the rqt_svo window shows "not connected" as well... what does it mean? Does ANYONE have ANY suggestions?  

I am guessing the problem is related to qt library... I am using ROS kinetic and had some troubles with the GUI stuff in general; kinetic uses Qt5 whereas previous versions use Qt4, so I replaced <QtGui> to <QtWidgets> in the svo.widget.py file. This enabled me to actually have the rqt window to pop up (yay!), however, I have no idea what is "not connected" and how to proceed... 

Thanks!

OHussain, 2018-05-15:
@astridpetitjean 
> kinetic uses Qt5 whereas previous versions use Qt4, so I replaced to in the svo.widget.py file.

Hey, I tried to find this file, but it doesn't seem to exist in either either the svo master or dependencies. Can I have a little more detail on this please?

astridpetitjean, 2018-05-17:
@OHussain 

Hello. I vaguely remember that I had to change some names due to change from Qt4 to Qt5 in one file (or more?)... but I don't remember what exactly, sorry. 

In the end I got it to work, but did not really need it... 

Sorry again for not being able to help you more... 

Best, 

Astrid

OHussain, 2018-05-17:
@astridpetitjean 
I figured it out. For other who may need this, in the file rpg_svo/rqt_svo/src/rqt_svo/svo_widget.py
you need to change the line:
`from python_qt_binding.QtGui import QWidget` 
to
`from python_qt_binding.QtWidgets import QWidget`
Because in QT5 QWidget is not part of QtGui. But as I also found out, this gui was not an important component. Thank you for your help though.

mukcit, 2018-12-11:
Hi, 
 I have the same problem like @clauniel, @ank700. so far ı couldnt find any solution. 
is there any one find it to help me?

thanks a lot.

/camera/image_raw
/clicked_point
/initialpose
/move_base_simple/goal
/rosout
/rosout_agg
/svo/dense_input
/svo/image
/svo/image/compressed
/svo/image/compressed/parameter_descriptions
/svo/image/compressed/parameter_updates
/svo/image/compressedDepth
/svo/image/compressedDepth/parameter_descriptions
/svo/image/compressedDepth/parameter_updates
/svo/image/theora
/svo/image/theora/parameter_descriptions
/svo/image/theora/parameter_updates
/svo/image_raw
/svo/info
/svo/keyframes
/svo/keyframes_array
/svo/points
/svo/points_array
/svo/pose
/svo/remote_key
/tf
/tf_static