uzh-rpg/rpg_svo #67: Segfaults on odroid-u3

clauniel, 2015-01-22:
SVO sometimes segfaults when there are not enough features to match. I have uploaded an strace here: http://sprunge.us/SQLA .

GhisHD, 2015-01-25:
Hi
May be you can check issue #59.

Thks

zhujimson, 2015-01-29:
Thank you so much, but after I changed map.cpp and frame_handler_mono.cpp, and then I catkin_make again. 

So I roslaunch test launch or live launch, and rosbag the airground_rig_s3_2013-03-18_21-38-48.bag, then I rosrun rviz rviz or rqt.

I still got nothing.Here are some pictures, please tell me how can I get through this, please.

zhujimson, 2015-01-29:
I can not upload any images here, I want to send you some images about my situation, Mr/Mrs GhisHD, it would be convenient if I can have your e-mail address.

GhisHD, 2015-01-29:
Hi Clauniel
Just to be sure about your issue:
You was face to a segment fault problem (crash)? Are you already be face to the same issue ?
Can you provide some call stack of your problem to have a more clear idea of the state of the program during the segment fault (with original code and with modified one if the call stack is different) ?

Hi zhujimson
Can you pleasr provide me your mail i will answer to it?

Thanks
Ghis

zhujimson, 2015-02-02:
Hi Ghis:
            Thanks for your feedback. This is my email: 632453964@qq.com. I would love to communicate with you guys about robot controlling.Thanks.

Your dear
Jimson