uzh-rpg/rpg_svo #78: Install error

alexandreiftm, 2015-03-18:
root@alexandre:~# cd workspace
root@alexandre:~/workspace# cd rpg_svo/svo
root@alexandre:~/workspace/rpg_svo/svo# cd build
root@alexandre:~/workspace/rpg_svo/svo/build# cmake ..
-- Eigen found (include: /usr/include/eigen3)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/alexandre/workspace/rpg_svo/svo/build
root@alexandre:~/workspace/rpg_svo/svo/build# make
[  4%] Building CXX object CMakeFiles/svo.dir/src/frame_handler_mono.cpp.o
cc1plus: error: unrecognized command line option ‘-mfpu=neon’
cc1plus: error: unrecognized command line option ‘-mfpu=neon’

/home/alexandre/workspace/rpg_svo/svo/src/frame_handler_mono.cpp:1:0: error: bad value (armv7-a) for -march= switch
make[2]: *** [CMakeFiles/svo.dir/src/frame_handler_mono.cpp.o] Error 1
make[1]: *** [CMakeFiles/svo.dir/all] Error 2
make: *** [all] Error 2

MuiLe, 2015-04-13:
Hi, 
You may should to run  export ARM_ARCHITECTURE=True

fickrie67, 2015-04-28:
Hi alexandreiftm,
Did you solve the problem? i would like to know because i have the same problem, even i ran export ARM_ARCHITECTURE=True 
Thank you

fickrie67, 2015-04-28:
Hi @alexandreiftm,
Did you solve the problem? i would like to know because i have the same problem, even i ran export ARM_ARCHITECTURE=True 
Thank you

cnsimbe, 2015-06-08:
Hi,
I had the same error.

I have previously installed svo_ros and the vikit, without the MSF package.
To correct this problem, I deleted the svo_ros and vikit source files and recloned them, and followed the procedure on Vision Estimation page (https://pixhawk.org/dev/ros/visual_estimation)

-- Basically delete your previous svo_ros catkin workspace and repeat the installation.