uzh-rpg/rpg_svo #79: the translation of camera remain -0.231303 0.226981 -0.327816 while testing "rpg_141122_borg_1__img.tar.gz" of your dataset

liqile, 2015-03-23:
I  tried to run "test_pipeline" with images in http://rpg.ifi.uzh.ch/datasets/libsvo/rpg_141122_borg_1__img.tar.gz 
I output the translation vector of the camera "vo_->lastFrame()->T_f_w_.translation()"
and found that after the frame #543, the translation remained "-0.231303 0.226981 -0.327816"
I didn't change the calibration of the camera, I only changed the "test_pipeline.cpp"
how could I do to make it work well, thanks in advance

Frame-Id: 543 	#Features: 0 	Proc. Time: 0.774ms 	-0.231303 0.226981 -0.327816
Frame-Id: 544 	#Features: 6 	Proc. Time: 2.6ms 	-0.231303 0.226981 -0.327816
Frame-Id: 545 	#Features: 3 	Proc. Time: 3.546ms 	-0.231303 0.226981 -0.327816
Frame-Id: 546 	#Features: 10 	Proc. Time: 4.631ms 	-0.231303 0.226981 -0.327816
Frame-Id: 547 	#Features: 5 	Proc. Time: 3.044ms 	-0.231303 0.226981 -0.327816
Frame-Id: 548 	#Features: 4 	Proc. Time: 3.086ms 	-0.231303 0.226981 -0.327816
Frame-Id: 549 	#Features: 8 	Proc. Time: 5.013ms 	-0.231303 0.226981 -0.327816
Frame-Id: 550 	#Features: 2 	Proc. Time: 7.108ms 	-0.231303 0.226981 -0.327816
Frame-Id: 551 	#Features: 3 	Proc. Time: 5.229ms 	-0.231303 0.226981 -0.327816
Frame-Id: 552 	#Features: 7 	Proc. Time: 3.751ms 	-0.231303 0.226981 -0.327816
Frame-Id: 553 	#Features: 5 	Proc. Time: 3.482ms 	-0.231303 0.226981 -0.327816
Frame-Id: 554 	#Features: 9 	Proc. Time: 4.04ms 	-0.231303 0.226981 -0.327816
Frame-Id: 555 	#Features: 8 	Proc. Time: 4.371ms 	-0.231303 0.226981 -0.327816
Frame-Id: 556 	#Features: 6 	Proc. Time: 4.408ms 	-0.231303 0.226981 -0.327816
Frame-Id: 557 	#Features: 8 	Proc. Time: 4.779ms 	-0.231303 0.226981 -0.327816
Frame-Id: 558 	#Features: 13 	Proc. Time: 4.694ms 	-0.231303 0.226981 -0.327816
Frame-Id: 559 	#Features: 5 	Proc. Time: 4.883ms 	-0.231303 0.226981 -0.327816
Frame-Id: 560 	#Features: 9 	Proc. Time: 4.493ms 	-0.231303 0.226981 -0.327816
Frame-Id: 561 	#Features: 9 	Proc. Time: 4.443ms 	-0.231303 0.226981 -0.327816
[WARN] Relocalizing frame
Frame-Id: 562 	#Features: 7 	Proc. Time: 4.514ms 	-0.231303 0.226981 -0.327816
Frame-Id: 563 	#Features: 4 	Proc. Time: 4.973ms 	-0.231303 0.226981 -0.327816
Frame-Id: 564 	#Features: 10 	Proc. Time: 4.908ms 	-0.231303 0.226981 -0.327816
Frame-Id: 565 	#Features: 12 	Proc. Time: 4.157ms 	-0.231303 0.226981 -0.327816
Frame-Id: 566 	#Features: 8 	Proc. Time: 4.966ms 	-0.231303 0.226981 -0.327816
Frame-Id: 567 	#Features: 8 	Proc. Time: 5.086ms 	-0.231303 0.226981 -0.327816
Frame-Id: 568 	#Features: 6 	Proc. Time: 4.642ms 	-0.231303 0.226981 -0.327816
Frame-Id: 569 	#Features: 3 	Proc. Time: 4.907ms 	-0.231303 0.226981 -0.327816
Frame-Id: 570 	#Features: 10 	Proc. Time: 5.325ms 	-0.231303 0.226981 -0.327816
[WARN] Not enough matched features.
Frame-Id: 571 	#Features: 2 	Proc. Time: 6.471ms 	-0.231303 0.226981 -0.327816
Frame-Id: 572 	#Features: 5 	Proc. Time: 4.931ms 	-0.231303 0.226981 -0.327816
Frame-Id: 573 	#Features: 5 	Proc. Time: 4.443ms 	-0.231303 0.226981 -0.327816

the last tracking of frames remains "-0.231303 0.226981 -0.327816"

cfo, 2015-03-23:
where did you get this dataset from?

liqile, 2015-03-23:
@cfo 
thanks for your reply
 I just visited your website http://rpg.ifi.uzh.ch/datasets/ and found some image data,  then I clicked and downloaded them..
I also try to run svo with image data in " nslam_test_data.tar.gz "
the tracking translation of frames remains "0 0 0" after the frame #679
May I know the calibration of these image data? thanks.

cfo, 2015-03-23:
sorry @liqile , we are not supporting these datasets. i suggest you use datasets from some other benchmarks such as the RGB-D dataset from TU Munich.
best,
christian

liqile, 2015-03-24:
@cfo 
thanks for your suggestion, I will try rgb-d dataset later.
but there was another strange question I have met. 
runing svo with same dataset might acquire different results.
 I recently ran your svo code with "nslam_test_data.tar.gz" for severial times
some times it worked very well while others it remains "0 0 0"

here are the results while working well
Frame-Id: 990 	#Features: 121 	Proc. Time: 7.229ms 	-0.494394 -0.37153 -0.110571
Frame-Id: 991 	#Features: 121 	Proc. Time: 6.121ms 	-0.471636 -0.36297 -0.113943
Frame-Id: 992 	#Features: 121 	Proc. Time: 6.606ms 	-0.453895 -0.355346 -0.114976
Frame-Id: 993 	#Features: 121 	Proc. Time: 6.635ms 	-0.435896 -0.347435 -0.116808
Frame-Id: 994 	#Features: 121 	Proc. Time: 6.331ms 	-0.421834 -0.341542 -0.117862
Frame-Id: 995 	#Features: 121 	Proc. Time: 6.595ms 	-0.406163 -0.339702 -0.11915
Frame-Id: 996 	#Features: 121 	Proc. Time: 6.882ms 	-0.381514 -0.337676 -0.120786
Frame-Id: 997 	#Features: 121 	Proc. Time: 5.861ms 	-0.364305 -0.335738 -0.122962
Frame-Id: 998 	#Features: 121 	Proc. Time: 6.859ms 	-0.348073 -0.332281 -0.125433
Frame-Id: 999 	#Features: 121 	Proc. Time: 6.27ms 	-0.32751 -0.329334 -0.125775
Frame-Id: 1000 	#Features: 121 	Proc. Time: 5.835ms 	-0.311631 -0.326861 -0.127597

and here are the results while acquire "0 0 0"
Frame-Id: 990 	#Features: 25 	Proc. Time: 10.713ms 	0 0 0
Frame-Id: 991 	#Features: 31 	Proc. Time: 13.027ms 	0 0 0
Frame-Id: 992 	#Features: 25 	Proc. Time: 12.496ms 	0 0 0
Frame-Id: 993 	#Features: 32 	Proc. Time: 12.876ms 	0 0 0
Frame-Id: 994 	#Features: 23 	Proc. Time: 12.999ms 	0 0 0
Frame-Id: 995 	#Features: 22 	Proc. Time: 12.618ms 	0 0 0
Frame-Id: 996 	#Features: 24 	Proc. Time: 12.347ms 	0 0 0
Frame-Id: 997 	#Features: 29 	Proc. Time: 12.286ms 	0 0 0
Frame-Id: 998 	#Features: 30 	Proc. Time: 12.12ms 	0 0 0
Frame-Id: 999 	#Features: 26 	Proc. Time: 12.288ms 	0 0 0
Frame-Id: 1000 	#Features: 23 	Proc. Time: 11.285ms 	0 0 0

cfo, 2015-03-24:
that's because svo lost tracking. i suggest to use the ROS version because there you get visual feedback of what's actually happening!

liqile, 2015-03-26:
@cfo 
hello, I am trying to run rgb-d dataset from tum.
but i don't know how to change the code to run the dataset