uzh-rpg/rpg_svo #80: error occurs while running rgb-d dataset from http://vision.in.tum.de

liqile, 2015-03-27:
hello, every one
I am running svo with "rgbd_dataset_freiburg1_xyz.bag"
I have changed the vo_node.cpp to set the camera and image stream topics and recompiled it
the topics are
 /camera/depth/image (sensor_msgs/Image) contains the depth map
/camera/rgb/camera_info (sensor_msgs/CameraInfo) contains the intrinsic camera parameters for the RGB camera, as reported by the OpenNI driver
I typed three commands in three terminals
$ roslaunch svo_ros test_rig3.launch
$ rosrun rviz rviz
$ rosbag play ~/rgbd_dataset_freiburg1_xyz.bag

but I saw nothing on rviz window

on the rosbag teminal, it stopped when arriving at duration of 3.780078 / 30.428861 
the rosbag terminal displayed as follows:
$ rosbag play ~/rgbd_dataset_freiburg1_xyz.bag 
[ INFO] [1427433073.617844147]: Opening /home/hongbohuang/rgbd_dataset_freiburg1_xyz.bag

Waiting 0.2 seconds after advertising topics... done.

Hit space to toggle paused, or 's' to step.
 [RUNNING]  Bag Time: 1305031102.162342   Duration: 3.780078 / 30.428861     
Done.

the roslaunch terminal displayed errors as follows:
[ INFO] [1427432783.385024768]: SVO initialized
[ WARN] [1427432783.385963897]: Cannot find value for parameter: /camera/rgb/camera_info, assigning default: /camera/rgb/image_color
[ INFO] [1427433082.645796917]: RESET
terminate called after throwing an instance of 'std::runtime_error'
  what():  Frame: provided image has not the same size as the camera model or image is not grayscale
[svo-2] process has died [pid 9925, exit code -6, cmd /home/hongbohuang/catkin_ws/devel/lib/svo_ros/vo __name:=svo __log:=/home/hongbohuang/.ros/log/0240709e-d43f-11e4-bcdc-bcaec584f5df/svo-2.log].
log file: /home/hongbohuang/.ros/log/0240709e-d43f-11e4-bcdc-bcaec584f5df/svo-2*.log

should I run svo with grayscale image? if so, how to get the grayscale image from TU-munich rgb-d  dataset ?

Thanks a lot
best

                                                                                                              liqile
thanks in advance..

sunghoon031, 2016-04-11:
Try like this: in the launch file

```
<launch>
<node name="image_proc" pkg="image_proc" type="image_proc" ns="camera"/>

<node pkg="svo_ros" type="vo" name="svo" clear_params="true" output="screen">

<!-- Camera topic to subscribe to -->
	<param name="cam_topic" value="/camera/image_mono" type="str" />
```

joeyzhouyue, 2016-10-18:
Hello, 
I would like to know if the suggestion from sunghoon031 works?
I am having the same problem. 
Thanks

Ankita-Singh-21, 2021-03-12:
I also encounter the same situation. This is how I solved it :
1) Convert your image to grayscale:
        frame = cv2.cvtColor(frame1, cv2.COLOR_BGR2GRAY)
2) Give the correct encoding to cv_bridge(in case you are using cv_bridge) :
        pub.publish(bridge.cv2_to_imgmsg(frame, "mono8"))
3) know your image size with  : (frame.shape) and give that frame shape to your camera model yaml file.

Hope this helps!!

vakh1375, 2023-02-17:
I try to reach "Fig. 14: Estimated trajectory and pointcloud of the TUM “fr2 desk” dataset." that is mentioned in second version of SVO. but there is not any calib.yaml file. I make a calib file with some information from TUM website. but it does not have any effect. runnging any of TUM bag files in RVIZ show a big error on motion detection and also running svo_benchmark does not plot any graph (i think its because of SVO error on convergence in this dataset). 
any one can help?