uzh-rpg/rpg_svo #81: live camera

YacratesWyh, 2015-04-11:
I faced such error while using my cam.

[INFO] [1424879295.497286408]: Init: KLT tracked 170 features
[ INFO] [1424879295.497506571]: Init: KLT 50.5493px average disparity.
FATAL Homography Initialization: This motion case is not implemented or is degenerate. Try again. [ INFO] [1424879295.505870505]: Init: Homography RANSAC 170 inliers.
[ WARN] [1424879295.505963104]: Cannot set scene depth. Frame has no point-observations!
[ INFO] [1424879295.506023094]: Init: Selected second frame, triangulated initial map.
[ WARN] [1424879295.543877898]: SparseImgAlign: no features to track!
[ WARN] [1424879295.544176578]: Not enough matched features.
[svo-1] process has died [pid 14813, exit code -11, cmd /home/yale/catkin_ws/devel/lib/svo_ros/vo __name:=svo __log:=/home/yale/.ros/log/e370ef42-bd03-11e4-a237-3423873a379f/svo-1.log].
log file: /home/yale/.ros/log/e370ef42-bd03-11e4-a237-3423873a379f/svo-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done 

Can anyone help~

jiapei100, 2015-05-03:
Same issue found in my case...
It seems SVO is not a well-functional code, but only for a student's demo.
There must be a commercial version I guess....

VishnuDuttSharma, 2016-06-21:
I followed this: [Visual Position Estimation](http://web.archive.org/web/20160322005117/http://pixhawk.org/dev/ros/visual_estimationl) for running live. Also, try changing parameters in config.h to get better tracking.