uzh-rpg/rpg_svo #83: Error with nullptr when install

lx492008, 2015-04-16:
I try to install the SVO,but when catkin_make I get this error:

/home/lx/catkin_ws/src/rpg_svo/svo/src/map.cpp: In member function ‘svo::FramePtr svo::Map::getClosestKeyframe(const FramePtr&) const’:
/home/lx/catkin_ws/src/rpg_svo/svo/src/map.cpp:135:12: error: could not convert ‘nullptr’ from ‘std::nullptr_t’ to ‘svo::FramePtr {aka boost::shared_ptr<svo::Frame>}’
[ 95%] Building CXX object rpg_svo/svo/CMakeFiles/svo.dir/src/matcher.cpp.o
make[2]: *** [rpg_svo/svo/CMakeFiles/svo.dir/src/map.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [rpg_svo/svo/CMakeFiles/svo.dir/all] Error 2
make: *** [all] Error 2
Invoking "make" failed
What happened?How can I solve this?Thanks!

MuiLe, 2015-04-18:
Hi,
Maybe it is compiler issue. I had the same error with gcc 4.6, but when upgrading to gcc 4.8, the error goes away.
you can find more info by googling  error: could not convert ‘nullptr’ from ‘std::nullptr_t’

Regards,

lx492008, 2015-04-21:
thanks for your reply,the problem has solved!  I had update the gcc.

Athria, 2015-06-03:
Hello, sorry for the trouble. I am having your same problem so I updated to gcc 4.8 but the errors remained. Did you do anything else (I downloaded and compiled everything again just in case, but to no use). Thank you in advance