uzh-rpg/rpg_svo #90: rpg_svo doesn't get new features

exFirefly, 2015-05-06:
I got issue with getting new features for svo on live camera. It works well on features from first frame. If I slowly move it through table, svo doesn’t acquire new features, it just losing features from first frame. If I go back to first frame, it will find lost features. I can relocalize svo at any part of my table and it find a lot of new features.  But it won't add new features after localizing at first frame. Can someone give me an idea what I should change to let it work with new features?

I'm using logitech webcam with tried it on 640x480 and 30 fps also tried it with raspicam without ros. But in both cases algorithm didn't get new features.

zestyfirehare, 2015-05-17:
I have the exact same problem. I'm also using a Logitech webcam (c615) and am running SVO with ROS Indigo on Ubuntu 14.04.

The tracking is great at first, I can see the correct estimated orientation in rviz and I can see the sparse depth map being generated ... but when I move the camera, no new features are tracked.

It seems like it's detecting new features because there are green dots which flash and then immediately disappear ... but these new features are definitely not tracked.

Btw, I've tried using other resolutions as well (1280x720) and experienced the same result.

Thanks so much for any help!

magcayaz, 2015-07-08:
Any updates to this problem?
I am observing the same issue.

yang3kui, 2015-07-22:
I have the same problem too. Can anybody help?

alejodosr, 2016-02-03:
Same problem here. I've tried the solution of changing init_min_disparity to 10 but it does not work either. Any other solution?

cssexpert217, 2017-01-30:
Hi @alejodosr, @zestyfirehare, @exFirefly, @magcayaz,

Did you guys find any solution to this problem as I'm facing the same issue.

Booooooosh, 2017-03-24:
Has anyone got the solution? I am having the same problem here: whenever I start, it detects feature points perfectly, but when I move the camera far away from the starting point, it loses the feature points.

maocong90, 2017-09-21:
@exFirefly How to implement svo with raspicam in pi2(or pi3)

IbrahimSquared, 2018-10-01:
@maocong90 You'll get much better results by streaming from the Pi using raspivid gstreamer to your PC since SVO could become computationally exhausting for the Pi, but that way you won't be able to go higher than ~30 FPS. 

@Booooooosh @cssexpert217 @alejodosr unless you are using the same camera setup used in the original SVO you won't be able to get as good a performance. Regardless, SVO 1.0 will only work if the camera is not so shaky and is directly looking downwards with a wide field of view, meaning in order to get good tracking with a rolling shutter camera that has narrow field of view, you have to hold the camera higher and make it directly look downwards, and move it really steadily!

Lucksong, 2019-11-14:
Hi,guys,did you solve this problem?My camera is davis240c and its resolution is 240x180. When I run svo1.0 with this camera thera are not any features and camera poses even in the first frame? Can anyone give some help?
Thanks very much!