uzh-rpg/rpg_svo #93: Installation error with gcc 4.8

Athria, 2015-06-05:
I installed the package a few months ago in an Ubuntu 12.04 virtual machine on a Macbook, with ros Hydro and didn't have any problems. 

Now I need to install it in an IIyama laptop with an i7, the same Ubuntu and ros distro as before, but I was running into Assembler errors like the one from issue#7 (https://github.com/uzh-rpg/rpg_svo/issues/7) so i updated to gcc 4.8; since it didn't solve the issue I proceded to apply the solution offered in that issue and removed -march=native from the three CMakeLists.txt.  That solved to assembler errors but problems like the one from issue#83 (https://github.com/uzh-rpg/rpg_svo/issues/83) started:

/home/ire/catkin_ws/src/rpg_svo/svo/src/map.cpp: In member function ‘svo::FramePtr svo::Map::getClosestKeyframe(const FramePtr&) const’:
/home/ire/catkin_ws/src/rpg_svo/svo/src/map.cpp:135:12: error: could not convert ‘nullptr’ from ‘std::nullptr_t’ to ‘svo::FramePtr {aka boost::shared_ptr<svo::Frame>}’ 

The solution given in that issue was to update to gcc 4.8 but I already did and verified it with:
$ gcc --version
gcc (Ubuntu 4.8.1-2ubuntu1~12.04) 4.8.1

Does anybody have an idea what might be wrong with my install or any way to solve the issue? the package is still working in my other computer, but I have to use it with this one.

The complete log for the first and second compilation errors can be found here: https://gist.github.com/Athria/b1014d56a994f7f12c8a

Thank you in advance.

YoshuaNava, 2015-06-06:
Hi,
Have you tried this? http://stackoverflow.com/questions/29125396/building-svo-package-on-raspberry-pi

If that doesn't work, try executing "catkin_make VERBOSE=1", to (hopefully) obtain more information about the problem

Best regards,
Yoshua

Athria, 2015-06-06:
Hello, the solution given on that thread completely solved the issue. Thank you for the information about the gcc parser, I hadn't run into a similar problem before, but it's nice to know the cause in case something similar happens again. 

Thank you very much!

closed 2015-06-06 by Athria

pangfumin, 2015-07-27:
yes, it also work for me.
http://stackoverflow.com/questions/29125396/building-svo-package-on-raspberry-pi