uzh-rpg/rpg_svo #94: problem with finding pose

PorterUno, 2015-06-12:
Hi I have a problem, I would like to print the position of the cam in the world coordinates. I put in the code:
if(vo_->lastFrame() != NULL)
    {
    	std::cout << "Frame-Id: " << vo_->lastFrame()->id_ << " \t"
                  << "#Features: " << vo_->lastNumObservations() << " \t"
                  << "Proc. Time: " << vo_->lastProcessingTime()*1000 << "ms \t"
		  << "Pose: "<< vo_->lastFrame()->T_f_w_.inverse().translation()<<std::endl;
    }

And I get the error:

ubuntu-gnome@ubuntu-gnome:~/rpg_svo/svo/bin$ ./test_pipeline
[INFO] SVO initialized
reading image /home/ubuntu-gnome/Datasets/sin2_tex2_h1_v8_d/img/frame_000002_0.png
[INFO] RESET
[INFO] Init: Selected first frame.
Frame-Id: 0 	#Features: 0 	Proc. Time: 49.924ms 	Pose: 0
0
0
[INFO] Init: KLT tracked 565 features
[INFO] Init: KLT 25.3823px average disparity.
Frame-Id: 1 	#Features: 0 	Proc. Time: 515.353ms 	Pose: 0
0
0
[INFO] Init: KLT tracked 543 features
[INFO] Init: KLT 50.8035px average disparity.
Segmentation fault (core dumped)

My question is: what is the problem? How can I print position?

zhujimson, 2015-12-16:
@PorterUno hello，can you describe your question more detailed？I want to talk to you about this topic.

valenbase, 2017-06-29:
@PorterUno i met the same problem , have you resolved it?