uzh-rpg/rpg_svo #98: sophus include error in rpg_vikit with Ros Indigo

Athria, 2015-06-28:
I was previously running rpg_svo on Ros Indigo (in ubuntu 14.04.2) and it worked well, but then I tried to install the lsd_slam package and it compiled with errors; I deleted everything related to lsd_slam and tried to compile again but I started getting errors like this one when applying catkin_make:

/home/irene2/catkin_ws/src/rpg_vikit/vikit_common/include/vikit/math_utils.h:14:24: fatal error: sophus/se3.h: No such file or directory
 #include <sophus/se3.h>

I deleted the four packages from the svo installation, and reinstalled them following the instructions, but the error remained. The full log can be found here https://gist.github.com/Athria/65a45a4445c68d17cdf2

I tried deleting the source and devel folders and the other packages i had installed, but the problem remained. I checked and the se3.h file is indeed under ~/workspace/Sophus/sophus/  but cannot figure out why it doesn't find the file.

Thank you in advance.

YoshuaNava, 2015-06-28:
Hello,
Are you using the catkinized or the rosmake version of LSD_SLAM?

Athria, 2015-06-29:
Hello, thank you for the answer. I was using the rosmake version of lsd_slam because the catkinized one gave me problems before, but I managed to fix them and now both packages compile well together. I think it was an issue with the rosbuild folder. 

Thank you!

closed 2015-06-29 by Athria

PerryLiustc, 2016-03-09:
Hi,I missed the same question about se3.h. could you tell me how to fix it?

hashten, 2016-05-06:
@Athria : Please post the fix so others can benefit from it as well.

hashten, 2016-05-06:
@PerryLiustc 
Finally, I managed to get it to compile. What I did was to check under 
~/.cmake/packages
There I had two folders, one "sophus" and one "Sophus". In the Sophus one which is where cmake looks for Sophus when compiling svo. In the Sophus folder, there are two files. If you open them you can see that one points to your LSD SLAM build and the other points to the Sophus you installed for SVO. I temporally moved the other file to another folder and removed the ~/catkin_ws/build folder and did catkin_make again.

However, when I run svo, I still get 
[svo-2] process has died [pid 3801, exit code -4, ....
as soon as I turn on the /camera/image_raw topic.

extr15, 2017-01-02:
just a tip: it seems Sophus/se3.h is Sophus/se3.hpp on my computer, so maybe you should modify it to se3.hpp.