uzh-rpg/rpg_svo #99: catkin_make shows error on raspbian wheezy

zuzu837, 2015-07-02:
I tried following the tutorials and I was able to make it work on my laptop running ubuntu 14.04 LTS and ros indigo, but when I tried builiding it on my raspberry pi 2 running raspbian, catkin_make ran into errors as following: 

Built target svo_msgs_generate_messages_lisp
[ 25%] [ 27%] [ 29%] Generating C++ code from svo_msgs/Feature.msg
Generating C++ code from svo_msgs/NbvTrajectory.msg
Generating C++ code from svo_msgs/Info.msg
[ 30%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/atan_camera.cpp.o
cc1plus: error: bad value (native) for -march switch
cc1plus: error: bad value (native) for -march switch
rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/build.make:54: recipe for target 'rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/atan_camera.cpp.o' failed
make[2]: *** [rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/atan_camera.cpp.o] Error 1
CMakeFiles/Makefile2:1434: recipe for target 'rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/all' failed
make[1]: *** [rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 32%] Generating C++ code from svo_msgs/DenseInput.msg
[ 32%] Built target svo_msgs_generate_messages_cpp
Makefile:123: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

I am not sure how to resolve them. Did anyone face the same issue.

zuzu837, 2015-07-02:
Wait, I solved this, I needed to follow the last comment on this post:  https://github.com/uzh-rpg/rpg_svo/issues/7

It builds but it doesn't run. It gives the following error when I run the launch files svo_ros:

core service [/rosout] found
process[svo-1]: started with pid [5025]
create vo_node
[ WARN] [1434640962.472374545]: Cannot find value for parameter: svo/publish_img_pyr_level, assigning default: 0
[ WARN] [1434640962.481953943]: Cannot find value for parameter: svo/publish_every_nth_img, assigning default: 1
[ WARN] [1434640962.490914600]: Cannot find value for parameter: svo/publish_every_nth_dense_input, assigning default: 1
[ INFO] [1434640962.513538504]: Found parameter: svo/publish_world_in_cam_frame, value: 0
[ WARN] [1434640962.524138721]: Cannot find value for parameter: svo/publish_map_every_frame, assigning default: 0
[ WARN] [1434640962.532065955]: Cannot find value for parameter: svo/publish_point_display_time, assigning default: 0
[ WARN] [1434640964.122029684]: Cannot find value for parameter: svo/publish_markers, assigning default: 1
[ WARN] [1434640964.132595683]: Cannot find value for parameter: svo/publish_dense_input, assigning default: 0
[ INFO] [1434640964.141977584]: Found parameter: svo/accept_console_user_input, value: 1
[svo-1] process has died [pid 5025, exit code -7, cmd /home/pi/catkin_ws/devel/lib/svo_ros/vo __name:=svo __log:=/home/pi/.ros/log/c39e1c62-15cd-11e5-9cc2-74da381ad27e/svo-1.log].
log file: /home/pi/.ros/log/c39e1c62-15cd-11e5-9cc2-74da381ad27e/svo-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

and on the raspberry pi it gives the error: 
"unhandled fault alignment exception (0x001) at 0x0000001d"

zuzu837, 2015-07-02:
well, nobody helped, but I changed the launch file parameter and it runs, but somehow nothing gets outputted in " rostopic echo /svo/pose" while " rostopic /svo/points" works fine!

zuzu837, 2015-07-02:
oh it publishes, but after 20 secs or so! I guess I need to have more patience! 
any idea why it would be this slow?

Wholemeal, 2015-07-28:
Hey Zuzu, what did you change in launch file to get it to work? Would really appreciate the help!

zuzu837, 2015-07-28:
add this line: <param name="accept_console_user_input" value="true" /> if you got the same error, try toggling to false if it doesn't work

Saums, 2016-04-22:
@zuzu837 @Wholemeal  @kartikmohta @eliasm 

We're unable to use the command : cmake .. for fast corner detection. We face the below error: 
c++ error: unrecognized command line '-mmme'
c++ error: unrecognized command line '-msse'

You've got it to run on pi, it'll be great if you can help.

cfo, 2016-04-22:
you should compile form ARM: https://github.com/uzh-rpg/rpg_svo/wiki/Installation:-General-for-ARM-processors

Saums, 2016-04-22:
@cfo Thanks. It solved the problem. Missed that instruction.

Saums, 2016-04-22:
@cfo One more doubt, which camera did you use? I have a Raspberry pi Camera 1.3, will that do?

cfo, 2016-04-22:
i don't know, you have to try. we are using high quality machine vision cameras as described in the paper.

sujunqin123, 2016-12-23:
when i run roslaunch svo_ros test_rig3.launch, occur this error, how should handler this problem?
create vo_node
[ WARN] [1482507134.097185391]: Cannot find value for parameter: svo/publish_img_pyr_level, assigning default: 0
[ WARN] [1482507134.107859461]: Cannot find value for parameter: svo/publish_every_nth_img, assigning default: 1
[ WARN] [1482507134.117069332]: Cannot find value for parameter: svo/publish_every_nth_dense_input, assigning default: 1
[ WARN] [1482507134.138511632]: Cannot find value for parameter: svo/publish_world_in_cam_frame, assigning default: 1
[ WARN] [1482507134.147913992]: Cannot find value for parameter: svo/publish_map_every_frame, assigning default: 0
[ WARN] [1482507134.157744699]: Cannot find value for parameter: svo/publish_point_display_time, assigning default: 0
[ WARN] [1482507135.110867126]: Cannot find value for parameter: svo/publish_markers, assigning default: 1
[ WARN] [1482507135.120468014]: Cannot find value for parameter: svo/publish_dense_input, assigning default: 0
[ WARN] [1482507135.130094776]: Cannot find value for parameter: svo/accept_console_user_input, assigning default: 1
pure virtual method called
terminate called without an active exception
[svo-2] process has died [pid 6570, exit code -6, cmd /home/pi/catkin_ws/devel/lib/svo_ros/vo __name:=svo __log:=/home/pi/.ros/log/f7d68c38-c924-11e6-bac2-8a59541eb190/svo-2.log].
log file: /home/pi/.ros/log/f7d68c38-c924-11e6-bac2-8a59541eb190/svo-2*.log

Zengqingfeng123, 2021-05-27:
> when i run roslaunch svo_ros test_rig3.launch, occur this error, how should handler this problem?
> create vo_node
> [ WARN] [1482507134.097185391]: Cannot find value for parameter: [...]

I guess you need "catkin_make" first to build the svo flie.