uzh-rpg/rpg_svo #102: DepthFilter::updateSeed function's details on the derivation

tanmengwen, 2015-07-06:
Hello,
https://github.com/uzh-rpg/rpg_svo/blob/master/svo/src/depth_filter.cpp#L309

seems like this function doesn't have actual formulate in paper "Video-based, Real-Time Multi View
Stereo". Also I can't understand the updateSeed function directly from the source code, could anyone explain the details of the derivation? 

Thank you!

cfo, 2015-07-08:
There is a supplementary material to that paper that shows the derivation. however, there is a typo that we needed to fix to make it work. Equation (19) in the supplementary material should be

1/s^2 = 1/sigma^2 + 1/tau^2.

and we also added a normalization step:

C1=C1/(C1+C2), C2=C2/(C1+C2)

tanmengwen, 2015-07-08:
Hi cfo,

Could you tell me the supplementary material or papers' name? It seems that I still couldn't find the exact reference paper about the derivation. 

Thanks a lot!

HeYijia, 2016-04-06:
@tanmengwen you can download the supplementary material from the website
[http://www.sciencedirect.com/science/article/pii/S0262885611000138 ](url)
[image]

focs, 2016-04-08:
@HeYijia the link doesn't seem to work for me. Am I the only one?

HeYijia, 2016-04-09:
@focs , I updated my comment. good luck!

whoiscris, 2017-06-27:
the supplementary formulae can be found on the author's page too: http://george-vogiatzis.org/publications/ivcj2010supp.pdf