uzh-rpg/rpg_svo #108: Doesn't seem to create new keyframes...

andsmith, 2015-07-14:
Hello, I'm trying to get SVO to work for my robot, but I'm having trouble with it.

I'm using a PSEye camera (640x480 / 60fps), wider angle setting, which I calibrated using ethzasl code as recommended to an error ~ 0.3.

It looks like it first identifies lots of good feature points, and tracks them successfully, but as they move out of view, new feature points are not identified/tracked, and I hypothesized this was because for some reason new keyframes were not being created (and the initial keyframe was moving out of view).   I am pretty sure I'm moving it more than 15% depth value which is supposed to trigger a new keyframe.  I've attached a sequence of images showing this behavior.  What do you think is going on?   Could it be something as simple as bad/misspecified calibration?

[image]
[image]
[image]
[image]

Thanks for your help!

GhostFlying, 2015-07-30:
I find this problem, too. It seems that this package only detect features at start only.

inspirit, 2015-07-30:
you are probably rotating camera without any translation this kind of movement is a killer for most monocular slam systems. try to shift camera instead of rotating it. also u can try to change GridSize in Config class to smth smaller like 20 or 15

zhujimson, 2015-12-09:
@inspirit，I got this problem too，and when I just moved the camera away from origin place a little bit(just 1 meter far)，then the SVO fail.......

wallarelvo, 2016-01-11:
I also have this issue. It also occurs with small translation

anl13, 2016-01-20:
Me too, it fails at the second keyframe.

alejodosr, 2016-02-03:
Same problem here.. any found solution?

anl13, 2016-02-07:
Hello!
After many tests, I found that, this problem arises when features are not enough. So, I try to make the environment in a mess, and svo runs well.
Although this ' solution ' may seems not a real solution, I don't have other methods now. Maybe we should modify svo itself to solve the problem totally.

with best wishes!

An Liang 
from Tsinghua Univ. of China

ginowu, 2016-04-01:
The initialization is very important, and also a good camera. I got a close result as the data set, using a ov7251 camera.