uzh-rpg/rpg_svo #111: Fast (Corner Detector) - Installation: Plain CMake (No ROS)

Radiomyme, 2015-09-11:
Hi everyone,

I'm working an a project of drone and I'm having an issue with my Raspberry Pi 2 B. When I'm trying to install "Fast - Corner Detector" as https://github.com/uzh-rpg/rpg_svo/wiki/Installation:-Plain-CMake-(No-ROS), I've this on screen : 

pi@raspberrypi ~/workspace/fast/build $ make

/usr/bin/cmake -H/home/pi/workspace/fast -B/home/pi/workspace/fast/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/pi/workspace/fast/build/CMakeFiles /home/pi/workspace/fast/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/pi/workspace/fast/build'
make -f CMakeFiles/fast.dir/build.make CMakeFiles/fast.dir/depend
make[2]: Entering directory '/home/pi/workspace/fast/build'
cd /home/pi/workspace/fast/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/pi/workspace/fast /home/pi/workspace/fast /home/pi/workspace/fast/build /home/pi/workspace/fast/build /home/pi/workspace/fast/build/CMakeFiles/fast.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/pi/workspace/fast/build'
make -f CMakeFiles/fast.dir/build.make CMakeFiles/fast.dir/build
make[2]: Entering directory '/home/pi/workspace/fast/build'
/usr/bin/cmake -E cmake_progress_report /home/pi/workspace/fast/build/CMakeFiles 1
[ 20%] Building CXX object CMakeFiles/fast.dir/src/fast_10.cpp.o
/usr/bin/c++   -Dfast_EXPORTS -DTEST_DATA_DIR=\"/home/pi/workspace/fast/test/data\" -Wall -Werror -Wno-unused-variable -Wno-unused-but-set-variable -Wno-unknown-pragmas -Wall -Werror -Wno-unused-variable -Wno-unused-but-set-variable -Wno-unknown-pragmas -O3 -mmmx -msse -msse -msse2 -msse3 -mssse3 -fomit-frame-pointer -fPIC -I/home/pi/workspace/fast/include -I/home/pi/workspace/fast/src -I/usr/local/include/opencv -I/usr/local/include    -o CMakeFiles/fast.dir/src/fast_10.cpp.o -c /home/pi/workspace/fast/src/fast_10.cpp
cc1plus: error: unrecognized command line option ‘-mmmx’
cc1plus: error: unrecognized command line option ‘-msse’
cc1plus: error: unrecognized command line option ‘-msse’
cc1plus: error: unrecognized command line option ‘-msse2’
cc1plus: error: unrecognized command line option ‘-msse3’
cc1plus: error: unrecognized command line option ‘-mssse3’
CMakeFiles/fast.dir/build.make:60: recipe for target 'CMakeFiles/fast.dir/src/fast_10.cpp.o' failed
make[2]: *** [CMakeFiles/fast.dir/src/fast_10.cpp.o] Error 1
make[2]: Leaving directory '/home/pi/workspace/fast/build'
CMakeFiles/Makefile2:66: recipe for target 'CMakeFiles/fast.dir/all' failed
make[1]: *** [CMakeFiles/fast.dir/all] Error 2
make[1]: Leaving directory '/home/pi/workspace/fast/build'
Makefile:119: recipe for target 'all' failed
make: *** [all] Error 2

I have Cmake installed (sudo apt-get cmake), Gcc (4.6.3) and OpenCV.

How can I solve the problem ?

Thank you for help

Martin

dwikholm, 2017-04-10:
I'm having this same issue now. Were you able to find a solution?

dwikholm, 2017-04-10:
I was able to figure it out based on the answer provided for the same question you posted on Stack Overflow:
http://stackoverflow.com/questions/32523589/svo-fast-coner-detector-installation-plain-cmake-no-ros