uzh-rpg/rpg_svo #115: Launch failes as libSophus.so cannot be found

nash911, 2015-09-23:
I get the following error when I try to Run SVO on the dataset. 

/.../SVO/Workspace_SVO$ roslaunch svo_ros test_rig3.launch
... logging to /home/nash/.ros/log/a8ca3ee0-6198-11e5-ad10-00255686fc1e/roslaunch-nashLaptop-16733.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nashLaptop:36559/

SUMMARY
========

CLEAR PARAMETERS
 * /svo/

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.10
 * /svo/cam_cx: 0.45905
 * /svo/cam_cy: 0.510056
 * /svo/cam_d0: 0.932
 * /svo/cam_fx: 0.509326
 * /svo/cam_fy: 0.796651
 * /svo/cam_height: 480
 * /svo/cam_model: ATAN
 * /svo/cam_topic: /camera/image_raw
 * /svo/cam_width: 752
 * /svo/grid_size: 30
 * /svo/init_rx: 3.14
 * /svo/init_ry: 0.0
 * /svo/init_rz: 0.0
 * /svo/loba_num_iter: 0
 * /svo/max_n_kfs: 10

NODES
  /
    svo (svo_ros/vo)

auto-starting new master
process[master]: started with pid [16745]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a8ca3ee0-6198-11e5-ad10-00255686fc1e
process[rosout-1]: started with pid [16758]
started core service [/rosout]
/home/nash/.../Workspace_SVO/devel/lib/svo_ros/vo: error while loading shared libraries: libSophus.so: cannot open shared object file: No such file or directory
process[svo-2]: started with pid [16776]
[svo-2] process has died [pid 16776, exit code 127, cmd /home/nash/Dropbox/PhD/Skycap/Project/SVO/Workspace_SVO/devel/lib/svo_ros/vo __name:=svo __log:=/home/nash/.ros/log/a8ca3ee0-6198-11e5-ad10-00255686fc1e/svo-2.log].
log file: /home/nash/.ros/log/a8ca3ee0-6198-11e5-ad10-00255686fc1e/svo-2*.log

I had to make install Sophus as catkin_make could not find it while compiling SVO project.

Any help would be highly appreciated.

nash911, 2015-09-23:
Solved.

Since I has installed Sophus, I had to run ldconfig as root to update the cache .

closed 2015-09-23 by nash911

wally1002, 2021-03-09:
Type in terminal :     sudo ldconfig    

then run the launch file