uzh-rpg/rpg_svo #119: Losing pose estimation when another plane comes into view

Wholemeal, 2015-10-20:
The following is part of my dataset:

https://www.dropbox.com/sh/3pluv689hdeug83/AABBeAUNMJl7eX-_xiiGcQGIa?dl=0

A camera flies over a building that obscures the ground plane. What are some of the parameters that will help capture the detail of the building or determine it's pose estimation? Any help would be very much appreciated!

cfo, 2015-10-20:
For the triangulation, the search range range along the epipolar line is limited by the min/max depth. 
You should be able to overcome such large depth discontinuities by setting the parameter 0.5 to a lower value here: https://github.com/uzh-rpg/rpg_svo/blob/master/svo/src/frame_handler_mono.cpp#L120

Wholemeal, 2015-10-20:
Hello Christian,

I have changed the parameter to 0.2 (added an extra config param in svo ros). It hasn't changed the results, here of what I see:

https://youtu.be/D0bl0PRfWDs

cfo, 2015-10-20:
what if you make it even smaller?

Wholemeal, 2015-10-20:
https://www.youtube.com/watch?v=34CasLz4vQw&feature=youtu.be

here it is at 0.02

cfo, 2015-10-20:
does is work when the camera is flying higher?

Wholemeal, 2015-10-20:
Yes! Looks good.

https://youtu.be/sfWjeDyNM18

If I were to go smaller would it eventually recognise the plane? Does adjusting the depth filter affect accuracies translations/rotations?

Thanks for your help!

cfo, 2015-10-20:
if you reduce the min-depth for the depth filter it will require more processing time to search the longer epipolar line, but it should get more robust for cases like yours.
How about changing the keyframe frequency by reducing Config::kfSelectMinDist

Wholemeal, 2015-10-20:
Two runs at 0.2 and 0.06
https://youtu.be/46euqnbTG6I

0.02 and 0.06
https://youtu.be/RUFq35TUjj4

Changliu52, 2016-05-13:
Hi @cfo 
I am trying to use a 170 degree camera with SVO and with 20 degree forward tilted.
Will the above method will make it more suitable to this setup?

trixr4kdz, 2018-02-09:
Hi, I was wondering if any of you guys can provide me with a list of parameters you used to be able to get the vision package working. I just need some example configuration that worked to see if there would be any improvements with respect to my camera. I am currently having my camera be around 5 ft from the ground, but the video from @Wholemeal shows that the feature detection works better when the camera is much higher, is the svo package more ideal for higher elevations?