uzh-rpg/rpg_svo #125: Estimate Trajectory from SVO not matching the Ground Truth for the sin2_tex2_h1_v8_d

sjl2, 2015-12-03:
Hello,

I am running SVO without ROS using the sin2_tex2_h1_v8_d data-set. By adding a couple lines to the test_pipeline code, I have the estimate trajectory being written for the test data-set. I have it working exactly like tracePose is used in the svo_ros BenchmarkNode class. The data is writing quite well, including the same exact format as the original ground-truth trajectory_nominal.txt file included.

The issue I'm running into is that when run with the provided data-set, the trajectory (x,y,z) does not match the ground truth provided. Is this expected? Unfortunately, the plots of the trajectories do not appear to even follow the same direction scaled or be off by translation. I have attached the found trajectories below. 

[image]

Thank you!

cfo, 2015-12-04:
as long as you don't set the initial position to the one of the ground-truth trajectory it will start estimating from the identity pose. furthermore, using a monocular camera, it is impossible to estimate the scale of the map, that's why the scale of the ground-truth and the estimate is different. in vikit_py there are some functions for the alignment (Sim3)

sjl2, 2015-12-04:
I see. Thank you for the fast response!

closed 2015-12-04 by sjl2