uzh-rpg/rpg_svo #134: UI 302 camera settings to make SVO work?

ank700, 2016-01-20:
Hello guys, I was able to run the dataset to confirm if SVO is correctly installed. 
Now, I want to use UI 302 camera with SVO. Can someone suggest me what parameters should I change to make it work properly.
I have prepared a calibration file in atan format. Now when I run SVO, i can see only random movements that too only if it is able to track some features otherwise nothing happens. I am using the rviz config file for the dataset.

Any help is much appreciated.
Thanks

gscortichini, 2016-02-09:
Let me know if you find any solutions

ank700, 2016-05-17:
Hi @gscortichini , sorry for replying after a very long time. I am still unable to make it work with UI cam even with a 4.5mm lens. Were you able to solve the problem?
I get the following when launching live.launch file:

[ INFO] [1463489249.975034966]: Init: KLT 25.8142px average disparity.
[ INFO] [1463489250.055628502]: Init: KLT tracked 134 features
[ INFO] [1463489250.055681675]: Init: KLT 26.0387px average disparity.
[ INFO] [1463489250.135194538]: Init: KLT tracked 134 features
[ INFO] [1463489250.135244947]: Init: KLT 28.3954px average disparity.
[ INFO] [1463489250.213666483]: Init: KLT tracked 134 features
[ INFO] [1463489250.213715920]: Init: KLT 39.1255px average disparity.
[ INFO] [1463489250.290888687]: Init: KLT tracked 134 features
[ INFO] [1463489250.290940644]: Init: KLT 48.9859px average disparity.
[ INFO] [1463489250.367922505]: Init: KLT tracked 134 features
[ INFO] [1463489250.367975837]: Init: KLT 56.6401px average disparity.
[ INFO] [1463489250.383453044]: Init: Homography RANSAC 90 inliers.
[ INFO] [1463489250.383567176]: Init: Selected second frame, triangulated initial map.
[ WARN] [1463489250.428254608]: Tracking less than 50 features!
[ WARN] [1463489250.428302177]: Lost 50 features!
[ WARN] [1463489250.504102433]: Relocalizing frame
[ WARN] [1463489250.584872093]: Not enough matched features.
[ WARN] [1463489250.973436117]: Tracking less than 50 features!
[ WARN] [1463489251.514019034]: Relocalizing frame
[ WARN] [1463489251.595076560]: Not enough matched features.
[ WARN] [1463489252.523886833]: Relocalizing frame
[ WARN] [1463489253.533820675]: Relocalizing frame
[ WARN] [1463489254.543648075]: Relocalizing frame
[ INFO] [1463489254.543702718]: No reference keyframe.
[ INFO] [1463489254.699036212]: No reference keyframe.
[ WARN] [1463489255.553571646]: Relocalizing frame
[ WARN] [1463489256.563414220]: Relocalizing frame
[ WARN] [1463489257.573350475]: Relocalizing frame

After this I have to exit the program because it keeps on giving the same error.

gscortichini, 2016-05-18:
in my case, i solved the problem with a better calibration software. First, i used PTAM calibration camera to get the atan model, but i had the same results as you.
Finally, i tried the ROS calibration camera and that made the trick! I supposed that software works better with my camera (PointGrey BlackFly 2.3 Mpx , 40fps) than PTAM. The only remarkable difference is the PINHOLE model that you get.

I see that you have a good framerate and a light camera resolution, so everything should be fine. I`m not an expert on lens, but in my opinion the only alternative reason of your issue could be your lens.

ank700, 2016-05-18:
Did you make any tests with the pose values? How are the final pose values? Is there a huge error?

I also calibrated my camera with cameracalibration.py. Can you, if possible, send me a rosbag file of the image stream and the pose values. Then, I can compare it with my own setup.
Thanks

ythej, 2016-05-23:
Hi @gscortichini ,

I have even tried running SVO with PointGrey BFLY-PGE but couldn't get any better results when compared with the results that I have got when I tried with the Microsoft LifeCam HD. Could you please share your camera launch file to let me check once again?

Did you face any problems with running rviz? After I installed FlyCam, rviz gives segmentation fault. I have tried all the solutions that mentioned on the website for this error but couldn't figure it out.

Thank you.

gscortichini, 2016-05-24:
Hi @ythej , this is my launch file:

<launch>
  
    <node pkg="svo_ros" type="vo" name="svo" clear_params="true" output="screen">
    
        
        <param name="cam_topic" value="/btr_bfly_camera_node/image_raw" type="str" />
        
        
        <rosparam file="$(find svo_ros)/param/camera_atan1.yaml" />
        
        
        <rosparam file="$(find svo_ros)/param/vo_fast.yaml" />
	<param name="init_rx" value="0.00" />
	<param name="init_ry" value="0.00" />
	<param name="init_rz" value="0.00" />
	<param name="max_fps" value="300" />
	<param name="quality_min_fts" value="10" />
    </node>
        
</launch>

I used the "btr-ros-pkg" ROS package (videomode 4 , pixelformat 0 ) to get frames from the camera. It works exactly like FlyCap2.

I had problems with rviz as you mentioned before, in my case the reason was the bad buffer size ubuntu used.  Then i found this solution, in order to increase the buffer size: 

sudo sysctl –w net.core.rmem_max=1048576 net.core.rmem_default=1048576

(if you google that command you can find better explanations to that issue...)

gscortichini, 2016-05-24:
Hi @ank700 , i don`t have any bag files and i don`t have my camera at the moment. I hope you`ll find a solution.

ythej, 2016-05-24:
Hi @gscortichini,

Thanks a lot for sharing the launch file.
It looks similar to mine and I even have set the framerate to 60fps, disabled the autofocus by setting it to zero, contrast, and brightness to avoid losing the features. But it didn't work, though.
It starts well in the beginning but will keep on loosing features immediately.

I even tried with Microsoft LifeCam using below launch file but the same thing repeats.

`<launch>
  <node name="camera" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video1" />
    <param name="framerate" value="30" />
    <param name="image_width" value="1280" />
    <param name="image_height" value="720" />
    <param name="pixel_format" value="mjpeg" />
    <param name="camera_frame_id" value="camera" />
    <param name="io_method" value="mmap"/>
    <param name="do_rectify" value="true"/>
    <param name="autofocus" value="false"/>
    <param name="focus" value="0" />
    <param name="contrast" value="32" />
    <param name='brightness' value="32" />
    <param name="camera_info_url" type="string" value="file:///home/c7031091/Documents/calibration/logitech/cal720.yml" />
  </node>
  <node name="image_view" pkg="image_view" type="image_view" respawn="false" output="screen">
    <remap from="image" to="/camera/image_raw"/>
    <param name="autosize" value="false" />
  </node>
</launch>` 

Please suggest if you have any suggestions :)
Thank you :)

cfo, 2016-05-25:
hi @ythej 
if you use double the image resolution than in the example, i suggest to increase the pyramid levels:
```
n_pyr_levels = 4
klt_max_level = 5
klt_min_level = 3
```

i think you also need to increase the max reprojection error:
```
reproj_thresh = 3 or 4
```

you can probably double the grid size
```
grid_size = 50
```

and to make it more robust, it's generally good to use more features:
```
max_fts = 200
```

ythej, 2016-05-25:
Hi @cfo,

Thanks a lot for the reply :)

As you told, this is my parameter file:

grid_size: 50
max_n_kfs: 50
loba_num_iter: 10
max_fts: 200
n_pyr_levels: 4
klt_max_level: 5
klt_min_level: 3
reproj_thresh: 4

And this is my launch file:

<launch>
  
    <node pkg="svo_ros" type="vo" name="svo" clear_params="true" output="screen">
    
        
        <param name="live_cam_topic" value="/camera/image_raw" type="str" />
        
        
        <rosparam file="$(find svo_ros)/param/calib_new.yaml" />
        
        
        <rosparam file="$(find svo_ros)/param/vo_accurate.yaml" />

	
    	<param name="init_rx" value="3.14" />
    	<param name="init_ry" value="0.00" />
    	<param name="init_rz" value="0.00" />

    </node>
        
</launch>

This my calibration file:

cam_model: Pinhole
cam_width: 1280
cam_height: 720
cam_fx: 1043.411950
cam_fy: 1036.666624
cam_cx: 598.396422
cam_cy: 388.282087
cam_d0: 0.046226
cam_d1: -0.068894
cam_d2: -0.000586
cam_d3: -0.012369

The results remain as usual. I made sure that camera is pointed downwards. I even tried to check the algorithm with the KITTI dataset but it is giving memory allocation error.

Could you please help in anyway?

Thank you.

ank700, 2016-06-07:
Hi @ythej , calibrate the camera again with the calibration tool at https://github.com/ethz-asl/ethzasl_ptam . It will give you a camera calibration in a slightly different format. I am sure this should work.
It helped in my case.

closed 2016-06-13 by ank700