uzh-rpg/rpg_svo #136: Not enough matched features

Syvashonok, 2016-02-04:
Hi. I have problem, error appears when the camera is moving

[ INFO] [1454592801.706759921]: Init: KLT 48.1078px average disparity.
[ INFO] [1454592801.731399581]: Init: KLT tracked 115 features
[ INFO] [1454592801.731654563]: Init: KLT 48.7893px average disparity.
[ INFO] [1454592801.761542092]: Init: KLT tracked 115 features
[ INFO] [1454592801.761625664]: Init: KLT 49.2004px average disparity.
[ INFO] [1454592801.800481154]: Init: KLT tracked 115 features
[ INFO] [1454592801.800552682]: Init: KLT 49.8124px average disparity.
[ INFO] [1454592801.845089826]: Init: KLT tracked 115 features
[ INFO] [1454592801.845170727]: Init: KLT 50.5734px average disparity.
[ INFO] [1454592801.848203823]: Init: Homography RANSAC 78 inliers.
2-View BA: Error before/after = 0.000349 / 0.000099
2-View BA: Wrong edges =  0
[ INFO] [1454592801.849447989]: Init: Selected second frame, triangulated initial map.
[ WARN] [1454592802.741079709]: Tracking less than 50 features!
[ WARN] [1454592802.790671594]: Relocalizing frame
[ INFO] [1454592802.793425049]: Relocalization successful.
[ INFO] [1454592802.916798718]: Relocalization successful.
[ INFO] [1454592803.260940256]: Relocalization successful.
[ WARN] [1454592803.282250410]: Tracking less than 50 features!
[ WARN] [1454592803.315918570]: Not enough matched features.
[ WARN] [1454592803.794247051]: Relocalizing frame
[ WARN] [1454592804.334439693]: Not enough matched features.
[ WARN] [1454592804.802048962]: Relocalizing frame
[ WARN] [1454592805.404971272]: Not enough matched features.
[ WARN] [1454592805.806456518]: Relocalizing frame
[ WARN] [1454592806.427680284]: Not enough matched features.
[ WARN] [1454592806.828181238]: Relocalizing frame
[ WARN] [1454592807.450965350]: Not enough matched features.
[ WARN] [1454592807.845509632]: Relocalizing frame
[ WARN] [1454592808.493078786]: Not enough matched features.
[ WARN] [1454592808.892221718]: Relocalizing frame
[ WARN] [1454592809.498456839]: Not enough matched features.
[ WARN] [1454592809.897183577]: Relocalizing frame
[ WARN] [1454592810.524472254]: Not enough matched features.
[ WARN] [1454592810.900385506]: Relocalizing frame
[ WARN] [1454592811.534541670]: Not enough matched features.
[ WARN] [1454592811.930092784]: Relocalizing frame
[ WARN] [1454592812.537803543]: Not enough matched features.
[ WARN] [1454592812.962529686]: Relocalizing frame
[ WARN] [1454592813.591117696]: Not enough matched features.
[ WARN] [1454592813.999831002]: Relocalizing frame
[ WARN] [1454592814.616662674]: Not enough matched features.
[ WARN] [1454592815.004774832]: Relocalizing frame
[ WARN] [1454592815.658301101]: Not enough matched features.
[ WARN] [1454592816.142703433]: Relocalizing frame
[ WARN] [1454592816.671875334]: Not enough matched features.
[ WARN] [1454592817.171043279]: Relocalizing frame
[ WARN] [1454592817.679810361]: Not enough matched features.
[ WARN] [1454592818.198586600]: Relocalizing frame
[ WARN] [1454592818.710290235]: Not enough matched features.
[ WARN] [1454592819.201020822]: Relocalizing frame
[ WARN] [1454592819.765903818]: Not enough matched features.
[ WARN] [1454592820.252453196]: Relocalizing frame
[ WARN] [1454592820.796263390]: Not enough matched features.
[ WARN] [1454592821.261044756]: Relocalizing frame

gscortichini, 2016-02-09:
i`m in the same situation..let me know if you find any solutions

alexbuyval, 2016-04-11:
Hi, I have same issue too. I tried to use the same video on other feature-base algorithms and they works well. Maybe some parameters are too restricted.

SergioGarG, 2016-07-18:
Hi, did you solve it?

zanazakaryaie, 2017-09-18:
Hi

I also got the same issue when running SVO on forward looking cameras such as Kitti or Euroc dataset.

They say that the keyframe selection strategy works for downward looking cameras. Forward motions generates large depths which crashes SVO. The strange thing is the SVO benchmarks revealed for these dataset!

Is there anybody who have solved this issue?

wwwefwehfrgjnrek, 2017-10-03:
Hey man,I have also encountered such a problem,but I was applied to the UAV. I modified the parameters according to your configuration, but the situation did not change. I just want to move a little, it will lose the feature points.I am very confused, when the repositioning situation appears, it seems that there are three parameters caused , which named img_align_n_tracked repr_n_mps repr_n_new_references,and I print out and found to have been 0. Can you help me, thanks!

kvmanohar22, 2019-05-11:
I have narrowed down the problem to [computing inliers](https://github.com/uzh-rpg/rpg_svo/blob/master/svo/src/initialization.cpp#L190). The reprojected points are not that close to the original points. I guess probably homography computation is flawed or camera extrinsics (R, t) computation is flawed (which does depend on computing homography.

EduardoSH94, 2020-03-26:
Same problem :(

zhangpeng66, 2021-12-08:
same problem ，how  to solve?

adamardiyanto, 2022-08-04:
same with mine