uzh-rpg/rpg_svo #137: Issue running the test dataset

w4zir, 2016-02-14:
the process dies when I run the bag file. Here is the log file:

usr/share/upstart/xdg:/etc/xdg', 'XDG_SEAT': 'seat0'}]
[roslaunch][INFO] 2016-02-14 21:42:18,609: process[svo-1]: args[[u'/home/khan/phd_ws/catkin_ws/devel/lib/svo_ros/vo', u'__name:=svo']]
[roslaunch][INFO] 2016-02-14 21:42:18,609: ... created process [svo-1]
[roslaunch.pmon][INFO] 2016-02-14 21:42:18,609: ProcessMonitor.register[svo-1]
[roslaunch.pmon][INFO] 2016-02-14 21:42:18,609: ProcessMonitor.register[svo-1] complete
[roslaunch][INFO] 2016-02-14 21:42:18,609: ... registered process [svo-1]
[roslaunch][INFO] 2016-02-14 21:42:18,609: process[svo-1]: starting os process
[roslaunch][INFO] 2016-02-14 21:42:18,609: process[svo-1]: start w/ args [[u'/home/khan/phd_ws/catkin_ws/devel/lib/svo_ros/vo', u'__name:=svo', u'__log:=/home/khan/.ros/log/49647eb8-d32e-11e5-bdd7-a0481ca4c0a1/svo-1.log']]
[roslaunch][INFO] 2016-02-14 21:42:18,609: process[svo-1]: cwd will be [/home/khan/.ros]
[roslaunch][INFO] 2016-02-14 21:42:18,612: process[svo-1]: started with pid [26047]
[roslaunch][INFO] 2016-02-14 21:42:18,612: ... successfully launched [svo-1]
[roslaunch][INFO] 2016-02-14 21:42:18,612: ... launch_nodes complete
[roslaunch.pmon][INFO] 2016-02-14 21:42:18,612: registrations completed <ProcessMonitor(ProcessMonitor-1, started daemon 140675407787776)>
[roslaunch.parent][INFO] 2016-02-14 21:42:18,612: ... roslaunch parent running, waiting for process exit
[roslaunch][INFO] 2016-02-14 21:42:18,612: spin
[roslaunch][ERROR] 2016-02-14 21:42:29,238: [svo-1] process has died [pid 26047, exit code -11, cmd /home/khan/phd_ws/catkin_ws/devel/lib/svo_ros/vo __name:=svo __log:=/home/khan/.ros/log/49647eb8-d32e-11e5-bdd7-a0481ca4c0a1/svo-1.log].
log file: /home/khan/.ros/log/49647eb8-d32e-11e5-bdd7-a0481ca4c0a1/svo-1*.log
[roslaunch.pmon][INFO] 2016-02-14 21:42:29,239: ProcessMonitor.unregister[svo-1] starting
[roslaunch.pmon][INFO] 2016-02-14 21:42:29,239: ProcessMonitor.unregister[svo-1] complete
[roslaunch][INFO] 2016-02-14 21:42:29,239: all processes on machine have died, roslaunch will exit
[roslaunch.pmon][INFO] 2016-02-14 21:42:29,339: ProcessMonitor._post_run <ProcessMonitor(ProcessMonitor-1, started daemon 140675407787776)>
[roslaunch.pmon][INFO] 2016-02-14 21:42:29,340: ProcessMonitor._post_run <ProcessMonitor(ProcessMonitor-1, started daemon 140675407787776)>: remaining procs are []
[roslaunch.pmon][INFO] 2016-02-14 21:42:29,342: ProcessMonitor exit: cleaning up data structures and signals
[roslaunch.pmon][INFO] 2016-02-14 21:42:29,343: ProcessMonitor exit: pmon has shutdown
[roslaunch][INFO] 2016-02-14 21:42:29,428: process monitor is done spinning, initiating full shutdown
[roslaunch][INFO] 2016-02-14 21:42:29,428: runner.stop()
[roslaunch][INFO] 2016-02-14 21:42:29,428: shutting down processing monitor...
[roslaunch][INFO] 2016-02-14 21:42:29,428: shutting down processing monitor <ProcessMonitor(ProcessMonitor-1, stopped daemon 140675407787776)>
[roslaunch.pmon][INFO] 2016-02-14 21:42:29,428: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 140675407787776)>
[roslaunch][INFO] 2016-02-14 21:42:29,429: ... shutting down processing monitor complete
[roslaunch][INFO] 2016-02-14 21:42:29,429: done
[roslaunch.pmon][INFO] 2016-02-14 21:42:29,429: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 140675407787776)>
[rospy.core][INFO] 2016-02-14 21:42:29,429: signal_shutdown [atexit]

hashten, 2016-05-03:
The exact same error happens to me both when trying the test dataset and with a live camera stream. Please let me know if you have found any solution.

saikiran321, 2017-04-06:
Hi, I got the same error, have you found any solution to this? @hashten @w4zir