uzh-rpg/rpg_svo #138: vo_ros/vo: malloc(): memory corruption caused by image size?

theclakuh, 2016-02-17:
Hi,

when I launch the svo_ros node `roslaunch svo_ros live.launch` I get an error:
```
[ INFO] [1455715165.624976791]: SVO initialized
[ INFO] [1455715165.625768093]: Found parameter: svo/cam_topic, value: /image_raw
[ INFO] [1455715197.445786640]: RESET
*** Error in `/home/user/workspace/indigo/devel/lib/svo_ros/vo': malloc(): memory corruption: 0x0000000001187920 ***
```
I would guess this is caused by a wrong image size configuration, so the OpenCV functions used by svo corrupt the memory working on the images.
I know there are others asking the same question, but didn't get answered yet, so I try to ask the questions by my self. I try to run the svo_ros node using, guess what? Yes, ROS. I calibrated my camera using the camera_calibration ros node. I translated the output to the .yaml-file representation in the <svo_ros_root>/param/camera_pinhole.yaml as follows
```
cam_model: Pinhole
cam_width: 644
cam_height: 484
cam_fx: 588.481298
cam_fy: 587.819899
cam_cx: 328.046456
cam_cy: 226.471844
cam_d0: -0.363501
cam_d1: 0.165011
cam_d2: 0.000571
cam_d3: -0.000577
```

As I got the image resolution 644x484 from the calibration data and confirmed it using the image_view node of the image_view package, I would say the image cam_width and height is I configured is correct. I even tried to set image_width and image_height in launch-filge to the same size, but it didn't change anything.
<launch>

    <node pkg="svo_ros" type="vo" name="svo" clear_params="true" output="screen">

        

        
        <param name="cam_topic" value="/image_raw" type="str" />

        
        <rosparam file="$(find svo_ros)/param/camera_pinhole.yaml" />

        
        <rosparam file="$(find svo_ros)/param/vo_fast.yaml" />

    </node>

</launch>

Did I miss to fit any parameters or other configurations or does someone have an other ideas what is wrong?

Thanks in advance.

amadoantonini, 2016-02-17:
Hi 

I actually found a solution to this issue. It was caused when making the image pyramid when the image had an odd number of columns.

I fixed it by modifying the file rpg_vikit/vikit_common/src/vision.cpp
Line 91 previously read:
  const int	stride  =  in.step.p[0];
I changed it to:
  const int	stride = in.step.p[0]/2*2;

Best

theclakuh, 2016-02-17:
I have no idea why, but it works. I hope this is pushed soon. Now I have to find out why there are not enough features found after the initialization of the second keyframe.

Thank you.

sujunqin123, 2017-02-05:
@amadoantonini Hi, I also get an error as follow, when I changed “const int stride = in.step.p[0];” to "const int stride = in.step.p[0]/2*2;", then do "catkin_make" command, this error also occurs, Any ideas? Thanks in advance.

*** Error in `/home/sujunqin/cakin_ws/devel/lib/svo_ros/vo': malloc(): memory corruption: 0x0000000001abb4f0 ***
[svo-2] process has died [pid 5553, exit code -6, cmd /home/sujunqin/cakin_ws/devel/lib/svo_ros/vo __name:=svo __log:=/home/sujunqin/.ros/log/91e3038a-ebb0-11e6-ba8c-4c3488f077a1/svo-2.log].
log file: /home/sujunqin/.ros/log/91e3038a-ebb0-11e6-ba8c-4c3488f077a1/svo-2*.log