uzh-rpg/rpg_svo #142: Result without ROS and trajectory are different

Nabetalion, 2016-03-16:
I try to use SVO without ROS. I follow the README.md and run the test_pipeline.cpp, but result are different.

I used the dataset: sin2_tex2_h1_v8_d.tar.gz. And, I compute it and got a result. The results in frame 185 are:
Trajectry:
3.680000 frame_000185 24.6517 2.8944 2.0000 0.8648 -0.0374 0.0000 0.0000
Computed result:
11.6264 4.94662 -0.028772
I guess we should obtain similar value between 24.6517 and 11.6264. The difference is double size. 

I add the following code shown below in "test_pipeline.cpp". This code is made from the "https://github.com/uzh-rpg/rpg_svo/wiki/Notation"
  // access the pose of the camera via vo_->lastFrame()->T_f_w_.
 Eigen::Vector3d p;
 p = vo_->lastFrame()->T_f_w_.inverse().translation();
 std::cout << "pose:" << p[0] <<"\t"<< p[1]<<"\t"<<p[2]<<std::endl;

My questions are:
- This result is true?
- How to obtain the position, posture and seed of map without ROS?
- Can we use other dataset for SVO without ROS?

jixiedaima, 2018-03-15:
/home/y700/Desktop/pic1.png
/home/y700/Desktop/pic2.pngMH_03/mav0$ 
I can get get trajectory information,but I can't use it to evaluate the result. Do you know why?

jixiedaima, 2018-03-15:
[image]
[image]
I can get get trajectory information,but I can't use it to evaluate the result. Do you know why?

Sho412, 2018-06-05:
I have same issues.

base file: test_pipeline.cpp
insert position: after vo_addImage(img,0.01*img_id);
insert code:
Vector3d p;
p=Vector3d(vo_->lastFrame()->T_f_w_.inverse().translation();

Computed result:
11.55644	4.921084	-0.008531 

Is there any solution?

Sho412, 2018-06-06:
I'd found the result without ros is same as the result which is mention in other issue.
https://github.com/uzh-rpg/rpg_svo/issues/125

Need to 
-Set the initial position to the one of the ground-truth trajectory.
-Set the scale.
or
-Alignment after calculation