uzh-rpg/rpg_svo #143: SVO Crashes after fast movement of the camera

ginowu, 2016-04-04:
This issue can be easily reproduced when fast moving the camera. If not moving fast, this issue will every time pop up after several minutes operation of SVO.  I am running SVO on an ARM board, with global shutter and wide FOV camera, the camera works under 60FPS, calibrated with PinHole model. 

Everything works fine except this crash, if you need more information, please let me know, thanks!

Detailed backtrace logs as below:

[ INFO] [1459743582.568163366]: Init: KLT tracked 181 features
[ INFO] [1459743582.568622265]: Init: KLT 50.8176px average disparity.
[ INFO] [1459743582.775399697]: Init: Homography RANSAC 108 inliers.
2-View BA: Error before/after = 0.002261 / 0.002261
2-View BA: Wrong edges =  0
[ INFO] [1459743582.787112021]: Init: Selected second frame, triangulated initial map.
[ WARN] [1459743611.638249973]: Lost 42 features!
[ WARN] [1459743611.648372631]: Relocalizing frame
[ INFO] [1459743611.662497000]: Relocalization successful.
[ WARN] [1459743611.737652525]: Lost 41 features!
[ INFO] [1459743611.763768587]: Relocalization successful.
[ WARN] [1459743613.762549586]: Lost 42 features!
[ WARN] [1459743613.773775509]: Relocalizing frame
[ INFO] [1459743613.788123106]: Relocalization successful.
[ WARN] [1459743613.863412742]: Lost 43 features!
[ INFO] [1459743613.889019801]: Relocalization successful.
[ WARN] [1459743614.063808865]: Lost 43 features!
[ WARN] [1459743614.093338782]: Lost 41 features!
[ INFO] [1459743614.145711997]: Relocalization successful.
[ WARN] [1459743614.211366366]: Lost 44 features!
[ INFO] [1459743614.244494664]: Relocalization successful.
[ WARN] [1459743614.281957636]: Lost 44 features!
[ WARN] [1459743614.311359638]: Lost 41 features!
[ INFO] [1459743614.342045266]: Relocalization successful.
[ WARN] [1459743614.592520420]: Lost 41 features!
[ INFO] [1459743614.615404603]: Relocalization successful.
[ WARN] [1459743615.267691521]: Lost 41 features!
[ WARN] [1459743615.279631340]: Relocalizing frame
[ INFO] [1459743615.295451985]: Relocalization successful.
[ WARN] [1459743615.649694262]: Lost 41 features!
[ INFO] [1459743615.679092253]: Relocalization successful.
[ WARN] [1459743615.752373223]: Lost 41 features!
[ WARN] [1459743615.777413207]: Lost 41 features!
[ INFO] [1459743615.797496966]: Relocalization successful.
[ WARN] [1459743627.076553125]: Lost 41 features!
[ WARN] [1459743627.081644819]: Relocalizing frame
[ INFO] [1459743627.102625490]: Relocalization successful.
[ WARN] [1459743627.168522250]: Lost 41 features!
[ WARN] [1459743627.197364468]: Lost 42 features!
[ WARN] [1459743627.226960945]: Lost 41 features!
[ WARN] [1459743627.262436553]: Lost 42 features!
[ WARN] [1459743627.291777878]: Lost 46 features!
[ WARN] [1459743627.319172254]: Lost 48 features!
[ WARN] [1459743627.345300919]: Lost 44 features!
[ INFO] [1459743627.371584739]: Relocalization successful.
[ WARN] [1459743627.386140247]: Lost 41 features!
[ WARN] [1459743627.418334811]: Lost 47 features!
[ INFO] [1459743627.451215197]: Relocalization successful.
[ WARN] [1459743627.471029843]: Lost 42 features!
[ WARN] [1459743627.496638413]: Lost 41 features!
[ INFO] [1459743627.527399456]: Relocalization successful.
[ WARN] [1459743627.618765621]: Lost 41 features!
[ INFO] [1459743627.662957449]: Relocalization successful.
[ WARN] [1459743636.413749661]: Lost 41 features!
[ WARN] [1459743636.420277479]: Relocalizing frame
[ INFO] [1459743636.453482913]: Relocalization successful.
[ WARN] [1459743665.863879871]: Lost 41 features!
[ WARN] [1459743665.871568765]: Relocalizing frame
[ INFO] [1459743665.897666233]: Relocalization successful.
[ WARN] [1459743666.448371880]: Lost 41 features!
[ WARN] [1459743666.508982834]: Lost 43 features!
[ INFO] [1459743666.583197177]: Relocalization successful.
[ WARN] [1459743676.873246847]: Tracking less than 50 features!
[ WARN] [1459743676.881718541]: Relocalizing frame
[... 63 lines cut ...]
$2 = (void (*)(void)) 0xb6f2b0be <svo::SparseImgAlign::computeResiduals(Sophus::SE3 const&, bool, bool)+678>
(gdb) p cur_img_ptr
$3 = (uint8_t *) 0x803ed865 <error: Cannot access memory at address 0x803ed865>
(gdb) p cur_img_ptr[0]
Cannot access memory at address 0x803ed865
(gdb) p patch_size_
$4 = 4
(gdb) p x
$5 = 0
(gdb) p cur_img.data
$6 = (uchar *) 0x3ed8e0 "\253w@m\201\247\304\274\315\322\355\374\374\375\370\372\375\376\375", '\376' <repeats 20 times>, "\375\327\206DZe\241\343\256\227\300\375\376\376\373\365\357\372\375\353\375", '\376' <repeats 20 times>, "\240k\\s\231\300\305\214p\206\270\345\375\375\375\371\372\357\335\375", '\376' <repeats 20 times>, "\202p\274\275\266\263\300\257fiu\204\243\320\366\374\375\361\361", '\376' <repeats 21 times>, "QT\300\317\315\314\273\236]Y]hs\201\224\302\360\376\361", '\376' <repeats 21 times>...
(gdb) p stride
$7 = 40
(gdb) p v_cur_i
$8 = <optimized out>
(gdb) p y
$9 = 0
(gdb) p patch_halfsize_[svo-1] killing on exit
Quit
(gdb) p patch_halfsize_
$10 = 2
(gdb) [svo-1] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done

fangthu, 2016-04-26:
@ginowu hi I think I also met with this problem. 
The reason maybe this :

When the camera is initialed or relocation, the algorithm capture the features in the picture. 
When the camera move fast, many features are no longer in the picture any more. So It lost many features. 
If the number of lost features is over than 40, the algorithm need to be relocated. 

Of course, you can change the parameter of 'quality_max_drop_fts' to smaller value. But in my opinion, this algorithm is not suitable when the scenario in picture changes fast.

zzaurak, 2017-05-05:
@ginowu have you found a solution yet?
same problem here, i'm using raspberry pi 3 and raspicam.
works kind of fine, but the occasional segfaults always come after fast movement of the camera...
gdb shows the same output, seems to have something to do with this line...
`/root/catkin_ws/src/rpg_svo/svo/src/sparse_img_align.cpp:209
209 const float intensity_cur = w_cur_tl_cur_img_ptr[0] + w_cur_tr_cur_img_ptr[1] + w_cur_bl_cur_img_ptr[stride] + w_cur_br_cur_img_ptr[stride+1];`
has anybody else a recomendation how to trace this problem down?

dujue, 2017-09-11:
@ginowu @zzaurak have you found a solution yet?

maocong90, 2017-09-19:
@ginowu @fangthu  @zzaurak have you found a solution yet?

maocong90, 2017-09-19:
@ginowu 
I also met with this problem.
I am running SVO on pi 3 or pi 2, with Logitech camera or sj7000 camera, but FPS less than 12 . In ad dition, I am using usb_cam to get the image stream. So I am eager to know how to improve the low fps. Can you tell me more about the ARM board?

maocong90, 2017-09-21:
@zzaurak How to implement svo with raspicam in pi2(or pi3) ?