uzh-rpg/rpg_svo #147: vikit_common instalation problem on ubuntu 16.04

NikitaLiashenko, 2016-05-10:
I am trying to install vikit_common using Installation: Plain CMake (No ROS) article,  but get such error:
cmake ..

-- The C compiler identification is GNU 5.3.1
-- The CXX compiler identification is GNU 5.3.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!

Does anybody know how to fix this?

hdino, 2016-05-18:
Did you set USE_ROS to false as suggested [here](https://github.com/uzh-rpg/rpg_svo/wiki/Installation:-Plain-CMake-%28No-ROS%29) ?

NikitaLiashenko, 2016-05-18:
yeah, I have done it, and after that I have changed instruction for vikit_common

closed 2016-05-18 by NikitaLiashenko