uzh-rpg/rpg_svo #152: fatal error: svo_msgs/Info.h: No such file or directory

hch661100, 2016-06-16:
I run SVO on Flight Snapdragon board, but compiling fails when:

[ 67%] Building CXX object rpg_svo/svo_ros/CMakeFiles/svo_visualizer.dir/src/visualizer.cpp.o
/root/catkin_ws/src/rpg_svo/svo_ros/src/visualizer.cpp:23:27: fatal error: svo_msgs/Info.h: No such file or directory
 #include <svo_msgs/Info.h>
                           ^
compilation terminated.
make[2]: *** [rpg_svo/svo_ros/CMakeFiles/svo_visualizer.dir/src/visualizer.cpp.o] Error 1
make[1]: *** [rpg_svo/svo_ros/CMakeFiles/svo_visualizer.dir/all] Error 2
make: *** [all] Error 2
make: INTERNAL: Exiting with 5 jobserver tokens available; should be 4!
Invoking "make -j4 -l4" failed

Help me :(

Ignotus, 2016-10-28:
Same issue, looks like headers have not been generated by the svo_msgs package in some reason

bhairavmehta95, 2017-02-08:
Does anyone have a solution to this? Even when I source the ROS setup file, my compilation always stops here.