uzh-rpg/rpg_svo #156: ERROR:Cannot find value for parameter: svo/cam_model

hit-john, 2016-07-15:
when I roslaunch the live.launch to start the SVO, I got a error like this:
[image]
And there are my parameters:
[image]
 Does anyone have the same problem like this?the error told me that  'what():  Camera model not correctly specified.' Does it mean that my camera is not suitable for this program,should I need to replace a camera? Thanks~

focs, 2016-07-18:
Camera model refers to `ATAN` camera or `PinHole` camera. I think the used model should be specified somewhere in the configuration. 

ps: no need to buy a new camera for now.

Abishekbabu, 2022-06-29:
I am facing the same problem, did you solve it?