uzh-rpg/rpg_svo #157: svo/image empty although tracking seems good

Rouno, 2016-07-15:
Hi everyone,

I'm trying to run SVO on ROS Kinectic and ended up compiling SVO on my system and running it with a live camera.
Rviz with svo config file is loaded, svo_rqt as well as svo itself. Tracking seems fine but nothing happens in rviz : no visualization of svo/image, no features, no transform neither (and  "#Features" value is always 0 in rqt_svo )
When I pause SVO (aka reset) svo/image publishes back my usb_cam/raw_image data stream.

Can anybody help me please ?

Thanks a lot

Rouno, 2016-07-16:
Went a bit further, 

frame in svo_ros Visualizer::publishMinimal remains NULL forever 

while rqt_svo is showing "RUNNING" and "GOOD TRACKING" when I 
`rosbag play airground_rig_s3_2013-03-18_21-38-48.bag`

Somebody help, it's so frustrating to get svo almost running and not see the end result in rviz

Rouno, 2016-07-18:
Found a solution for this : I made last_frame_  public in svo/frame_handler_mono.h instead of protected

Somehow the function lastFrame() always returned NULL otherwise.

xiaodaxia, 2016-07-21:
I Had the same issue when using bag files from rpg_svo.  Maybe using kinetic is the origin of errors.
I also tracked to the frame always NULL problem, so no information is published to the "/image" topic, so rviz have nothing to show.