uzh-rpg/rpg_svo #158: live.launch segfault after initialization

nizarAbak-kali, 2016-07-15:
Hello i've been trying to make the SVO work for days... and i can't seem to understand why it crash every time i launch it .
I am on Xubuntu 14.04 , ROS indigo and using the IDS USB 3.0 uEye camera . I've calibrated my cam using the ethzasl ptam . 
here is a screenshot of the calibration file :
[image]
 
Here is a screenshot of the initialisation: 
[image]

and here a screenshot of the backtrack on gdb:
[image]

It seems that it's the method "addImage()" that  is problematic.
Thanks in advance for your reply.

focs, 2016-07-18:
Looks like OpenCV is having problems to copy the image. Can you make sure that the input is not empty (i.e. print `img.size()` in `addImagre()`)?

Does it crash with the first incoming image? or a after a few? is it a segmentation fault?

nizarAbak-kali, 2016-07-18:
It crash at begining of the launch i don't even see an image.

here is the screenshot of the print:
[image]

as you can see below on the screen, the image is not empty. Right after i got a segfault.

focs, 2016-07-18:
Then I'm not sure what the problem could be. I don't have much experience with ROS, but I see references in the bt to opencv2 and opencv3, could that be a problem?

nizarAbak-kali, 2016-07-18:
with more prints i found out that the segfault happen in the svo::FrameHandlerMono::addImage method at :

> new_frame_.reset(new Frame(cam_, img.clone(), timestamp));

focs, 2016-07-19:
That was my guess too. But if `img` is valid, I don't see why it should crash while `clone`ing.

Rouno, 2016-08-02:
Hi nizar, I have the problem on kinetic and Ubuntu Xenial. Have you managed to solve it yet ?

Best regards

nizarAbak-kali, 2016-08-03:
i've given up... (mostly because i've a lot of different opencv version and to not break something on my pc i prefered to postpone this task)

what is your opencv version ?

Rouno, 2016-08-03:
I have a stock ROS Kinetic so it should come with OpenCV 3 afaik

nizarAbak-kali, 2016-08-03:
i have a colleague with whom it worked and he had opencv 2.4.8 (don't know if it will help you)