uzh-rpg/rpg_svo #166: Failed to run test_pipeline

sogetibordes, 2016-09-20:
Hello guys,
I am descovering svo and I have some trouble running test_pipeline. When I enter "./test_pipeline", I get  this:

[INFO] SVO initialized
reading image /home/nico/Datasets/sin2_tex2_h1_v8_d/img/frame_000002_0.png
test_pipeline: /home/nico/Téléchargements/rpg_svo_svo_test_test_pipeline.cpp:70: 
 void svo::BenchmarNode::runFromFolder(): Assertion '!img.empty()' failed.
Abandon (core dumped)

Could you help me with it ?

Best regards

Nico

yang778, 2016-09-21:
Maybe the image path is incorrect. Try to use absolute path instead.

sogetibordes, 2016-09-21:
Thanks a lot Yang, that was it !

Lucky8Young, 2017-03-10:
Hello guys,
There was no problem when I followed the "Installation Instructions" of "No ROS (plain CMake)" on my Ubuntu16.04 computer.But when I followed the "Run Instructions" of "Run without ROS", after I entered "./test_pipeline", I got something:

terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid

I don't know what's wrong.
Could you please help me with it?
Thank you very much!