uzh-rpg/rpg_svo #168: Rviz No image recieved

karanchawla, 2016-09-24:
System: i7 -4720HQ 3.6Ghz
OS: Ubuntu 14.04 LTS with ROS Indigo. 

Everything compiled successfully . (Eventually) And the rosbag example seems to be working fine, i.e. features are getting detected, ros topics are echoing back the data. 

Only thing I can't see is the pose visualization in Rviz or the camera feed. I have attached the screen shots that may provide more information. 

Thank you. 
[image]
[image]

karanchawla, 2016-10-07:
`<launch>

  <node pkg="svo_ros" type="vo" name="svo" clear_params="true" output="screen">

	
	<param name="cam_topic" value="/image_mono" type="str" />
	
	
	<rosparam file="$(find svo_ros)/param/camera_pinhole.yaml" />

	
	<rosparam file="$(find svo_ros)/param/vo_robust.yaml" />
	
	
    <param name="init_rx" value="3.14" />
    <param name="init_ry" value="0.00" />
    <param name="init_rz" value="0.00" />

  </node>

</launch>`

Using this launch script solves the aforementioned problem.

closed 2016-10-07 by karanchawla

Ignotus, 2016-10-29:
Hey there, seems like I have the same issue. Where can I find the launch script?