uzh-rpg/rpg_svo #170: How to integrate SVO with IMU?

yang778, 2016-09-26:
Hi, all,
I'm trying to integrate SVO with IMU on UAV, but I'm not sure how to setup SVO with IMU. @cfo provided a link before(https://pixhawk.org/dev/ros/visual_estimation), but there is no content on that page any more. Does anyone know how to access the content in the above link or how to fuse SVO with IMU?
Thanks in advance!

karanchawla, 2016-10-05:
You can get the state of the camera from using this package and after applying the right rotation matrices, derive the pose of your UAV. You can then fuse it with your IMU using EKF or any other filter of your choice.

Booooooosh, 2017-05-10:
You can try to use the package here (http://wiki.ros.org/ethzasl_sensor_fusion), it fuses the output of the svo with the output of the imu. But still the initialization, that is the camera imu translation, the scale...is very crucial (actually I myself is struggling on that).