uzh-rpg/rpg_svo #171: ROS Kinetic: test_rig3.launch with example data doesn't work

KahnShi, 2016-10-05:
Hi all,

I am Ubuntu 16.04 64bits and ROS Kinetic.
I compiled and try to run test_rig3.launch with example data given by their package.
However, the result is same as #41 
The terminal is keeping waiting in  

> [INFO] [1408385938.551716596]: Init: Selected second frame, triangulated initial map

And /SVO/Image do not have any output.

Is there anyone have similar situation?

btw, I tried to enable/disable g2o, but it doesn't help.

Best,
Kahn

KahnShi, 2016-10-05:
My output in terminal is the following information:
`SUMMARY
========

CLEAR PARAMETERS
 * /svo/

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.5
 * /svo/cam_cx: 0.45905
 * /svo/cam_cy: 0.510056
 * /svo/cam_d0: 0.932
 * /svo/cam_fx: 0.509326
 * /svo/cam_fy: 0.796651
 * /svo/cam_height: 480
 * /svo/cam_model: ATAN
 * /svo/cam_topic: /camera/image_raw
 * /svo/cam_width: 752
 * /svo/grid_size: 30
 * /svo/init_rx: 3.14
 * /svo/init_ry: 0.0
 * /svo/init_rz: 0.0
 * /svo/loba_num_iter: 0
 * /svo/max_n_kfs: 10

NODES
  /
    svo (svo_ros/vo)

auto-starting new master
process[master]: started with pid [31451]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to cdcb7f2c-8b01-11e6-936e-c85b765648c0
process[rosout-1]: started with pid [31464]
started core service [/rosout]
process[svo-2]: started with pid [31482]
create vo_node
[ WARN] [1475675054.972821044]: Cannot find value for parameter: svo/publish_img_pyr_level, assigning default: 0
[ WARN] [1475675054.973379253]: Cannot find value for parameter: svo/publish_every_nth_img, assigning default: 1
[ WARN] [1475675054.973707317]: Cannot find value for parameter: svo/publish_every_nth_dense_input, assigning default: 1
[ WARN] [1475675054.974460837]: Cannot find value for parameter: svo/publish_world_in_cam_frame, assigning default: 1
[ WARN] [1475675054.974800522]: Cannot find value for parameter: svo/publish_map_every_frame, assigning default: 0
[ WARN] [1475675054.975114333]: Cannot find value for parameter: svo/publish_point_display_time, assigning default: 0
[ WARN] [1475675055.070796862]: Cannot find value for parameter: svo/publish_markers, assigning default: 1
[ WARN] [1475675055.071261355]: Cannot find value for parameter: svo/publish_dense_input, assigning default: 0
[ WARN] [1475675055.071789665]: Cannot find value for parameter: svo/accept_console_user_input, assigning default: 1
[ INFO] [1475675055.072221343]: Found parameter: svo/cam_model, value: ATAN
[ INFO] [1475675055.072622056]: Found parameter: svo/cam_width, value: 752
[ INFO] [1475675055.072980126]: Found parameter: svo/cam_height, value: 480
[ INFO] [1475675055.073456498]: Found parameter: svo/cam_fx, value: 0.509326
[ INFO] [1475675055.073828534]: Found parameter: svo/cam_fy, value: 0.796651
[ INFO] [1475675055.074193539]: Found parameter: svo/cam_cx, value: 0.45905
[ INFO] [1475675055.074558971]: Found parameter: svo/cam_cy, value: 0.510056
[ INFO] [1475675055.074965451]: Found parameter: svo/cam_d0, value: 0.932
[ WARN] [1475675055.075320145]: Cannot find value for parameter: svo/init_tz, assigning default: 0
[ WARN] [1475675055.075675278]: Cannot find value for parameter: svo/init_ty, assigning default: 0
[ WARN] [1475675055.076052683]: Cannot find value for parameter: svo/init_tx, assigning default: 0
[ INFO] [1475675055.076460539]: Found parameter: svo/init_rz, value: 0
[ INFO] [1475675055.076866884]: Found parameter: svo/init_ry, value: 0
[ INFO] [1475675055.077241458]: Found parameter: svo/init_rx, value: 3.14
[ WARN] [1475675055.077659810]: Cannot find value for parameter: svo/trace_name, assigning default: svo
[ WARN] [1475675055.077996076]: Cannot find value for parameter: svo/trace_dir, assigning default: /tmp
[ WARN] [1475675055.078303761]: Cannot find value for parameter: svo/n_pyr_levels, assigning default: 3
[ WARN] [1475675055.078644382]: Cannot find value for parameter: svo/use_imu, assigning default: 0
[ WARN] [1475675055.078990342]: Cannot find value for parameter: svo/core_n_kfs, assigning default: 3
[ WARN] [1475675055.079361483]: Cannot find value for parameter: svo/map_scale, assigning default: 1
[ INFO] [1475675055.079724265]: Found parameter: svo/grid_size, value: 30
[ WARN] [1475675055.080086332]: Cannot find value for parameter: svo/init_min_disparity, assigning default: 50
[... 53 lines cut ...]
[ INFO] [1475675089.635316608]: Init: KLT tracked 149 features
[ INFO] [1475675089.635346484]: Init: KLT 32.5355px average disparity.
[ INFO] [1475675089.669167068]: Init: KLT tracked 148 features
[ INFO] [1475675089.669196705]: Init: KLT 35.1747px average disparity.
[ INFO] [1475675089.702127691]: Init: KLT tracked 148 features
[ INFO] [1475675089.702156858]: Init: KLT 37.2156px average disparity.
[ INFO] [1475675089.736001551]: Init: KLT tracked 146 features
[ INFO] [1475675089.736049354]: Init: KLT 39.0457px average disparity.
[ INFO] [1475675089.768697197]: Init: KLT tracked 146 features
[ INFO] [1475675089.768744868]: Init: KLT 41.2904px average disparity.
[ INFO] [1475675089.802820754]: Init: KLT tracked 146 features
[ INFO] [1475675089.802865830]: Init: KLT 43.3591px average disparity.
[ INFO] [1475675089.835818567]: Init: KLT tracked 146 features
[ INFO] [1475675089.835844390]: Init: KLT 45.0894px average disparity.
[ INFO] [1475675089.869882364]: Init: KLT tracked 145 features
[ INFO] [1475675089.869940600]: Init: KLT 46.5785px average disparity.
[ INFO] [1475675089.902919172]: Init: KLT tracked 143 features
[ INFO] [1475675089.902966428]: Init: KLT 48.3242px average disparity.
[ INFO] [1475675089.936239467]: Init: KLT tracked 143 features
[ INFO] [1475675089.936300198]: Init: KLT 49.9558px average disparity.
[ INFO] [1475675089.969605017]: Init: KLT tracked 143 features
[ INFO] [1475675089.969668139]: Init: KLT 51.8553px average disparity.
[ INFO] [1475675089.971399201]: Init: Homography RANSAC 137 inliers.
[ INFO] [1475675089.971583836]: Init: Selected second frame, triangulated initial map.
`

yang778, 2016-10-12:
The "X" and "Y" in the last line of rviz_config.rviz are too large. For example, make X = 2 and Y = 2.

rehearsal-, 2016-12-08:
@KahnShi , same issue, always haunt at

> [ INFO] [1475675089.971583836]: Init: Selected second frame, triangulated initial map.

Please let me know if you have found any solution.

mkorkmazeem, 2017-02-22:
Same problem I faced. Nothing appears on Rviz.
Is it due to ROS Kinetic or anything else?
Any suggestions appreciated.
thx.

arjmacedo, 2017-02-22:
I've managed to make it work by downgrading the version of Eigen3 to the version 3.2.10 before compiling. 
This was commented here : https://github.com/uzh-rpg/rpg_svo/issues/174.