uzh-rpg/rpg_svo #173: Failed to run svo

vincentsogetibordes, 2016-10-11:
Hello all,

I am discovering SVO and i'm trying to install it(with ROS Kinetic).

When i type "catkin_make", i have this error :

[ 41%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/math_utils.cpp.o
In file included from /home/pi/catkin_ws/src/rpg_vikit/vikit_common/include/vikit/omni_camera.h:15:0,
                 from /home/pi/catkin_ws/src/rpg_vikit/vikit_common/src/omni_camera.cpp:10:
/home/pi/catkin_ws/src/rpg_vikit/vikit_common/include/vikit/math_utils.h:14:24: fatal error: sophus/se3.h: Aucun fichier ou dossier de ce type
 #include <sophus/se3.h>
                        ^
compilation terminated.
In file included from /home/pi/catkin_ws/src/rpg_vikit/vikit_common/include/vikit/atan_camera.h:20:0,
                 from /home/pi/catkin_ws/src/rpg_vikit/vikit_common/src/atan_camera.cpp:14:
/home/pi/catkin_ws/src/rpg_vikit/vikit_common/include/vikit/math_utils.h:14:24: fatal error: sophus/se3.h: Aucun fichier ou dossier de ce type
 #include <sophus/se3.h>
                        ^
compilation terminated.
In file included from /home/pi/catkin_ws/src/rpg_vikit/vikit_common/src/math_utils.cpp:8:0:
/home/pi/catkin_ws/src/rpg_vikit/vikit_common/include/vikit/math_utils.h:14:24: fatal error: sophus/se3.h: Aucun fichier ou dossier de ce type
 #include <sophus/se3.h>
                        ^
compilation terminated.
[ 41%] Built target svo_msgs_generate_messages_eus
Scanning dependencies of target svo_msgs_generate_messages
[ 41%] Built target svo_msgs_generate_messages
[ 43%] Building CXX object rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/vision.cpp.o
rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/build.make:100: recipe for target 'rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/math_utils.cpp.o' failed
make[2]: *** [rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/math_utils.cpp.o] Error 1
make[2]: *** Attente des tâches non terminées....
rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/build.make:77: recipe for target 'rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/omni_camera.cpp.o' failed
make[2]: *** [rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/omni_camera.cpp.o] Error 1
rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/build.make:54: recipe for target 'rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/atan_camera.cpp.o' failed
make[2]: *** [rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/src/atan_camera.cpp.o] Error 1
CMakeFiles/Makefile2:1808: recipe for target 'rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/all' failed
make[1]: *** [rpg_vikit/vikit_common/CMakeFiles/vikit_common.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Could I have some help with this? Thank you in advance!

sujunqin123, 2016-12-23:
I am also on Raspberry PI 3 Raspbian Jessie. But when i type "catkin_make", i have this error :catkin_make
Base path: /home/pi/ros_catkin_ws
Source space: /home/pi/ros_catkin_ws/src
Build space: /home/pi/ros_catkin_ws/build
Devel space: /home/pi/ros_catkin_ws/devel
Install space: /home/pi/ros_catkin_ws/install
####
#### Running command: "cmake /home/pi/ros_catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/pi/ros_catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/pi/ros_catkin_ws/install -G Unix Makefiles" in "/home/pi/ros_catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/pi/ros_catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/pi/ros_catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
CMake Warning (dev) at catkin/CMakeLists.txt:5 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    catkin_VERSION
This warning is for project developers.  Use -Wno-dev to suppress it.

-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 197 packages in topological order:
-- ~~  - genmsg
-- ~~  - gencpp
-- ~~  - genlisp
-- ~~  - genpy
-- ~~  - angles
-- ~~  - bond_core (metapackage)
-- ~~  - cmake_modules
-- ~~  - class_loader
-- ~~  - common_msgs (metapackage)
-- ~~  - common_tutorials (metapackage)
-- ~~  - convex_decomposition
-- ~~  - cpp_common
-- ~~  - desktop (metapackage)
-- ~~  - diagnostics (metapackage)
-- ~~  - eigen_stl_containers
-- ~~  - executive_smach (metapackage)
-- ~~  - geometry (metapackage)
-- ~~  - geometry_tutorials (metapackage)
-- ~~  - gl_dependency
-- ~~  - ivcon
-- ~~  - media_export
-- ~~  - message_generation
-- ~~  - message_runtime
-- ~~  - mk
-- ~~  - nodelet_core (metapackage)
-- ~~  - octomap (plain cmake)
-- ~~  - orocos_kdl (plain cmake)
-- ~~  - pr2_description
-- ~~  - python_orocos_kdl (plain cmake)
-- ~~  - qt_dotgraph
-- ~~  - qt_gui
-- ~~  - qt_gui_py_common
-- ~~  - qwt_dependency
-- ~~  - random_numbers
-- ~~  - robot (metapackage)
-- ~~  - robot_model (metapackage)
[... 150 lines cut ...]
-- ~~  - rqt_rviz
-- ~~  - rviz_plugin_tutorials
-- ~~  - rviz_python_tutorial
-- ~~  - urdf_tutorial
-- ~~  - vikit_common
-- ~~  - vikit_py
-- ~~  - vikit_ros
-- ~~  - svo
-- ~~  - svo_ros
-- ~~  - visualization_marker_tutorials
-- ~~  - xacro
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:63 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/pi/ros_catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/ros_catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

Have you meet that error? Thank you in advance!

AlexisTM, 2017-03-11:
@sujunqin123 

```
This workspace contains non-catkin packages in it, and catkin cannot build
a non-homogeneous workspace without isolation. Try the
'catkin_make_isolated' command instead.
```

You can try `catkin_make_isolated`