uzh-rpg/rpg_svo #174: ROS Kinetic: ubuntu 16.04 example doesn't work

mrpiccolo, 2016-10-19:
HI 
I am Ubuntu 16.04 64bits and ROS Kinetic.
I compiled both with and without G2O and try to run test_rig3.launch with example data given by their package.
But nothing appears in rviz and writing rostopic echo /svo/image or /svo/points 

this my output terminal:

SUMMARY
========

CLEAR PARAMETERS
 * /svo/

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.5
 * /svo/cam_cx: 0.45905
 * /svo/cam_cy: 0.510056
 * /svo/cam_d0: 0.932
 * /svo/cam_fx: 0.509326
 * /svo/cam_fy: 0.796651
 * /svo/cam_height: 480
 * /svo/cam_model: ATAN
 * /svo/cam_topic: /camera/image_raw
 * /svo/cam_width: 752
 * /svo/grid_size: 30
 * /svo/init_rx: 3.14
 * /svo/init_ry: 0.0
 * /svo/init_rz: 0.0
 * /svo/loba_num_iter: 0
 * /svo/max_n_kfs: 10

NODES
  /
    svo (svo_ros/vo)

auto-starting new master
process[master]: started with pid [24647]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2b33ab78-9622-11e6-8a36-382c4a156e0b
process[rosout-1]: started with pid [24660]
started core service [/rosout]
process[svo-2]: started with pid [24676]
create vo_node
[ WARN] [1476898418.469145291]: Cannot find value for parameter: svo/publish_img_pyr_level, assigning default: 0
[ WARN] [1476898418.469972971]: Cannot find value for parameter: svo/publish_every_nth_img, assigning default: 1
[ WARN] [1476898418.470645191]: Cannot find value for parameter: svo/publish_every_nth_dense_input, assigning default: 1
[ WARN] [1476898418.472220871]: Cannot find value for parameter: svo/publish_world_in_cam_frame, assigning default: 1
[ WARN] [1476898418.472905389]: Cannot find value for parameter: svo/publish_map_every_frame, assigning default: 0
[ WARN] [1476898418.473624601]: Cannot find value for parameter: svo/publish_point_display_time, assigning default: 0
[ WARN] [1476898418.631800043]: Cannot find value for parameter: svo/publish_markers, assigning default: 1
[ WARN] [1476898418.632324631]: Cannot find value for parameter: svo/publish_dense_input, assigning default: 0
[ WARN] [1476898418.632848917]: Cannot find value for parameter: svo/accept_console_user_input, assigning default: 1
[ INFO] [1476898418.633442144]: Found parameter: svo/cam_model, value: ATAN
[ INFO] [1476898418.634002801]: Found parameter: svo/cam_width, value: 752
[ INFO] [1476898418.634552071]: Found parameter: svo/cam_height, value: 480
[ INFO] [1476898418.635098428]: Found parameter: svo/cam_fx, value: 0.509326
[ INFO] [1476898418.635611587]: Found parameter: svo/cam_fy, value: 0.796651
[ INFO] [1476898418.636125383]: Found parameter: svo/cam_cx, value: 0.45905
[ INFO] [1476898418.636637511]: Found parameter: svo/cam_cy, value: 0.510056
[ INFO] [1476898418.637186937]: Found parameter: svo/cam_d0, value: 0.932
[ WARN] [1476898418.637756320]: Cannot find value for parameter: svo/init_tz, assigning default: 0
[ WARN] [1476898418.638295676]: Cannot find value for parameter: svo/init_ty, assigning default: 0
[ WARN] [1476898418.638819413]: Cannot find value for parameter: svo/init_tx, assigning default: 0
[ INFO] [1476898418.639338969]: Found parameter: svo/init_rz, value: 0
[ INFO] [1476898418.639855285]: Found parameter: svo/init_ry, value: 0
[ INFO] [1476898418.640371523]: Found parameter: svo/init_rx, value: 3.14
[ WARN] [1476898418.641002128]: Cannot find value for parameter: svo/trace_name, assigning default: svo
[ WARN] [1476898418.641565015]: Cannot find value for parameter: svo/trace_dir, assigning default: /tmp
[... 67 lines cut ...]
[ INFO] [1476898444.188357442]: Init: KLT tracked 146 features
[ INFO] [1476898444.188430842]: Init: KLT 41.2904px average disparity.
[ INFO] [1476898444.221987326]: Init: KLT tracked 146 features
[ INFO] [1476898444.222055087]: Init: KLT 43.3591px average disparity.
[ INFO] [1476898444.255318291]: Init: KLT tracked 146 features
[ INFO] [1476898444.255390743]: Init: KLT 45.0894px average disparity.
[ INFO] [1476898444.289329467]: Init: KLT tracked 145 features
[ INFO] [1476898444.289394234]: Init: KLT 46.5785px average disparity.
[ INFO] [1476898444.322727368]: Init: KLT tracked 143 features
[ INFO] [1476898444.322792549]: Init: KLT 48.3242px average disparity.
[ INFO] [1476898444.355824221]: Init: KLT tracked 143 features
[ INFO] [1476898444.355892949]: Init: KLT 49.9558px average disparity.
[ INFO] [1476898444.389832517]: Init: KLT tracked 143 features
[ INFO] [1476898444.389896650]: Init: KLT 51.8553px average disparity.
[ INFO] [1476898444.391765466]: Init: Homography RANSAC 137 inliers.
2-View BA: Error before/after = 0.000580 / 0.000580
2-View BA: Wrong edges =  0
[ INFO] [1476898444.392943445]: Init: Selected second frame, triangulated initial map.

without g2o i haven't 
2-View BA: Error before/after = 0.000580 / 0.000580
2-View BA: Wrong edges =  0
but the result is the same

any suggestion?

mrpiccolo, 2016-10-19:
i have installed opencv 3.1, it can generate issue?

tiagojdias, 2016-11-05:
I had the same issue. 

For me the solution was pretty simple, i've just made the downgrade of both ros (to indigo) and Ubuntu (to 14.04). 
I really don't know where is the problem. But at least i can say that everything is working fine now.

arjmacedo, 2016-12-29:
I was able to make it run in Ubuntu 16.
The problem was the new version of Eigen3 that does not work with the current version of SVO.
In order to solve it, we can download Eigen 3.2.10 and install it. After that, all we need to to is delete the previously built version of SVO and the CMake files created by catkin and rebuild it. After rebuilding, it should work. If you need more precise instructions, keep reading:
In order to install Eigen 3.2.10, download http://bitbucket.org/eigen/eigen/get/3.2.10.tar.bz2, then run:
```
tar -xf [name of the file]
cd [name of the file]
mkdir build
cd build
cmake ..
sudo make install
```
These steps might be simplified as Eigen3 is a bunch of headers and don't need to be compiled but i haven't tried a simplified version so, just to be safe, do all the steps above. They are not complicated anyways.
Then remove the folders build and devel from the catkin workspace and remove the CMakeLists.txt from the src folder. Assuming your catkin folder is called catkin_ws, it would be something like
```
cd catkin_ws
rm -r devel
rm -r build
rm src/CMakeLists.txt
catkin_make
```

The last command will rebuild the project and now it should work. When running catkin_make, check if the Eigen version found by cmake is the one in /usr/local/include. If it is different than that, e.g. /usr/include, then it is probably using the newer version and it shouldn't work.
Let me know if this solution works.

melchiorharcourt, 2017-01-04:
Great ! I will try this at once.

Kostech, 2017-01-16:
Thank you @arjmacedo, I had the same issue and it fixed the problem to me as well.

brzzda, 2017-03-31:
Not working for me. ROS kinetic kame, Ubuntu 16.04. I have installed Eigen3.2.10 but the output is still the same:
......
[ INFO] [1490965771.915452722]: Init: Homography RANSAC 143 inliers.
[ INFO] [1490965771.933345061]: Init: Selected second frame, triangulated initial map.

And thats it - nothing happening in rviz. 
Any suggestions will be highly appreciated.

danielgusland, 2017-04-30:
Not working for me etiher, however, I used version 3.2.8, available at:
http://bitbucket.org/eigen/eigen/get/3.2.8.tar.bz2

Also; Make sure you remove eigen before you install. In my case I did this:

- Remove Eigen
- Make sure it doesn't build
- Install eigen 3.2.8
- Delete devel and build and rebuild

Hope that works for you guys as well!

kilmarnock, 2017-05-15:
for me, arjmacedo's version does the trick. ubuntu 16.04 
I want to say that I had crashes before. The console went simply to values above 50:
`Init: KLT 50.1635px average disparity`
and then 
`[svo-1] process has died`
which was not easy to debug.

We should promote that 
# SVO does not work on ubuntu 16.04 without these steps
definetly, we should

githubwys, 2021-12-27:
great 
cmakelist.txt
it works. svo need the eigen3.2.10
but i still wonder why the svo-cmakelist.txt file finds the /usr/local/include/eigen3.2.10 first, not the /usr/include/eigen3.3.4.

and if i make another project , where cmakelist.txt wil find eigen first, will it make a mistake cause camkelist.txt find the eigen3.2.10 first , not the eigen3.3.4 that i truly need.