uzh-rpg/rpg_svo #185: [svo-2] process has died when Run SVO on a Dataset

sujunqin123, 2017-01-20:
svo_ros crashes when i run roslaunch svo_ros test_rig3.launch, rosrun rviz rviz -d ~cakin_ws/src/svo_ros/rviz_config.rviz and rosbag play airground_rig_s3_2013-03-18_21-38-48.bag.

The output message:
sujunqin@sujunqin:~$ roslaunch svo_ros test_rig3.launch
... logging to /home/sujunqin/.ros/log/5613b250-df17-11e6-97d8-4c3488f077a1/roslaunch-sujunqin-3722.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sujunqin:57056/

SUMMARY
========

CLEAR PARAMETERS
 * /svo/

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /svo/cam_cx: 0.45905
 * /svo/cam_cy: 0.510056
 * /svo/cam_d0: 0.932
 * /svo/cam_fx: 0.509326
 * /svo/cam_fy: 0.796651
 * /svo/cam_height: 480
 * /svo/cam_model: ATAN
 * /svo/cam_topic: /camera/image_raw
 * /svo/cam_width: 752
 * /svo/grid_size: 30
 * /svo/init_rx: 3.14
 * /svo/init_ry: 0.0
 * /svo/init_rz: 0.0
 * /svo/loba_num_iter: 0
 * /svo/max_n_kfs: 10

NODES
  /
    svo (svo_ros/vo)

auto-starting new master
process[master]: started with pid [3734]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5613b250-df17-11e6-97d8-4c3488f077a1
process[rosout-1]: started with pid [3747]
started core service [/rosout]
process[svo-2]: started with pid [3759]
create vo_node
[ WARN] [1484920200.790391439]: Cannot find value for parameter: svo/publish_img_pyr_level, assigning default: 0
[ WARN] [1484920200.791062070]: Cannot find value for parameter: svo/publish_every_nth_img, assigning default: 1
[ WARN] [1484920200.791599184]: Cannot find value for parameter: svo/publish_every_nth_dense_input, assigning default: 1
[ WARN] [1484920200.793744377]: Cannot find value for parameter: svo/publish_world_in_cam_frame, assigning default: 1
[ WARN] [1484920200.794745427]: Cannot find value for parameter: svo/publish_map_every_frame, assigning default: 0
[ WARN] [1484920200.795712673]: Cannot find value for parameter: svo/publish_point_display_time, assigning default: 0
[ WARN] [1484920200.961592271]: Cannot find value for parameter: svo/publish_markers, assigning default: 1
[ WARN] [1484920200.962739268]: Cannot find value for parameter: svo/publish_dense_input, assigning default: 0
[ WARN] [1484920200.963849960]: Cannot find value for parameter: svo/accept_console_user_input, assigning default: 1
[ INFO] [1484920200.965065778]: Found parameter: svo/cam_model, value: ATAN
[ INFO] [1484920200.966093980]: Found parameter: svo/cam_width, value: 752
[ INFO] [1484920200.967017556]: Found parameter: svo/cam_height, value: 480
[ INFO] [1484920200.967964738]: Found parameter: svo/cam_fx, value: 0.509326
[ INFO] [1484920200.968925633]: Found parameter: svo/cam_fy, value: 0.796651
[ INFO] [1484920200.969876267]: Found parameter: svo/cam_cx, value: 0.45905
[ INFO] [1484920200.970840636]: Found parameter: svo/cam_cy, value: 0.510056
[ INFO] [1484920200.971785425]: Found parameter: svo/cam_d0, value: 0.932
[ WARN] [1484920200.973390971]: Cannot find value for parameter: svo/init_tz, assigning default: 0
[ WARN] [1484920200.974923471]: Cannot find value for parameter: svo/init_ty, assigning default: 0
[ WARN] [1484920200.976596159]: Cannot find value for parameter: svo/init_tx, assigning default: 0
[ INFO] [1484920200.978030209]: Found parameter: svo/init_rz, value: 0
[... 65 lines cut ...]
[ INFO] [1484920254.302953803]: Init: KLT tracked 147 features
[ INFO] [1484920254.303042500]: Init: KLT 35.0074px average disparity.
[ INFO] [1484920254.358199373]: Init: KLT tracked 147 features
[ INFO] [1484920254.358273331]: Init: KLT 37.212px average disparity.
[ INFO] [1484920254.370096487]: Init: KLT tracked 147 features
[ INFO] [1484920254.370176654]: Init: KLT 39.0935px average disparity.
[ INFO] [1484920254.403020771]: Init: KLT tracked 147 features
[ INFO] [1484920254.403079975]: Init: KLT 41.0359px average disparity.
[ INFO] [1484920254.440462266]: Init: KLT tracked 147 features
[ INFO] [1484920254.440538358]: Init: KLT 43.2432px average disparity.
[ INFO] [1484920254.470725972]: Init: KLT tracked 147 features
[ INFO] [1484920254.470776944]: Init: KLT 45.2503px average disparity.
[ INFO] [1484920254.505241221]: Init: KLT tracked 146 features
[ INFO] [1484920254.505296345]: Init: KLT 47.631px average disparity.
[ INFO] [1484920254.540945213]: Init: KLT tracked 145 features
[ INFO] [1484920254.541016805]: Init: KLT 49.2348px average disparity.
[ INFO] [1484920254.572938850]: Init: KLT tracked 145 features
[ INFO] [1484920254.572997900]: Init: KLT 50.8822px average disparity.
[ INFO] [1484920254.636565022]: Init: Homography RANSAC 141 inliers.
vo: malloc.c:2372: sysmalloc: Assertion `(old_top == (((mbinptr) (((char *) &((av)->bins[((1) - 1) * 2])) - __builtin_offsetof (struct malloc_chunk, fd)))) && old_size == 0) || ((unsigned long) (old_size) >= (unsigned long)((((__builtin_offsetof (struct malloc_chunk, fd_nextsize))+((2 *(sizeof(size_t))) - 1)) & ~((2 *(sizeof(size_t))) - 1))) && ((old_top)->size & 0x1) && ((unsigned long) old_end & pagemask) == 0)' failed.
[svo-2] process has died [pid 3759, exit code -6, cmd /home/sujunqin/cakin_ws/devel/lib/svo_ros/vo __name:=svo __log:=/home/sujunqin/.ros/log/5613b250-df17-11e6-97d8-4c3488f077a1/svo-2.log].
log file: /home/sujunqin/.ros/log/5613b250-df17-11e6-97d8-4c3488f077a1/svo-2*.log

I'm using ubuntu 14.04LTS with ROS Indigo. 64-bits in thinkpad e460.
How can i solve this problem?

sujunqin123, 2017-02-07:
when I cmake without g2o, it run fine. I want to know that the problem result from g2o version?

xyh2211, 2024-04-25:
这个问题最终如何解决的？