uzh-rpg/rpg_svo #195: Extension to automatic camera exposure

Changliu52, 2017-03-28:
In comparison to recent DSO algorithm, one of the biggest drawbacks for SVO is that it is very sensitive to camera exposure changes (flickering).

I wonder is there anyone who can possibly give some hints here for possible code improvement to enable exposure invariant tracking for SVO?

Many thinks

eivan, 2017-03-28:
see in paper " http://rpg.ifi.uzh.ch/docs/TRO16_Forster-SVO.pdf ":
"More elaborate photometric modeling as proposed in DSO [42] may help SVO to cope with the abrupt illumination changes"
Have you considered this or similar?

Changliu52, 2017-03-28:
Thanks Eivan. That is exactly the paper, which inspired me to try to improve SVO.
Any suggestion on where to start from, to actually implement this to SVO?
I can see in the SVO code https://github.com/uzh-rpg/rpg_svo/blob/master/svo/src/feature_alignment.cpp#L288
This looks like somewhere we need to pay attention.