uzh-rpg/rpg_svo #197: Not able to run test files with ros

saikiran321, 2017-04-04:
Hi I'm using ROS indigo. I have tried to run SVO with both usb camera  and Dataset but there was an error.
**Error Details**
[ INFO] [1491319062.530179338]: Found parameter: svo/max_n_kfs, value: 10

[ WARN] [1491319062.530774573]: Cannot find value for parameter: svo/img_imu_delay, assigning default: 0
[ WARN] [1491319062.531440665]: Cannot find value for parameter: svo/max_fts, assigning default: 120
[ WARN] [1491319062.532605659]: Cannot find value for parameter: svo/quality_min_fts, assigning default: 50
[ WARN] [1491319062.534422005]: Cannot find value for parameter: svo/quality_max_drop_fts, assigning default: 40
[ INFO] [1491319062.534697655]: SVO initialized
[ INFO] [1491319062.536049583]: Found parameter: svo/cam_topic, value: /camera/image_raw
[ INFO] [1491319094.470399205]: RESET
[svo-1] process has died [pid 22614, exit code -11, cmd /home/sai/ros_packages/svo/devel/lib/svo_ros/vo __name:=svo __log:=/home/sai/.ros/log/149e036e-192f-11e7-a5bd-2c6e85ae6ca8/svo-1.log].
log file: /home/sai/.ros/log/149e036e-192f-11e7-a5bd-2c6e85ae6ca8/svo-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

I have tried to debug the issue. I don't know what went wrong as no error were displayed. One thing I observed was that  `roslaunch svo_ros test_rig3.launch`  fails as soon as I run the below command

> rosbag  play airground_rig_s3_2013-03-18_21-38-48.bag

saikiran321, 2017-04-06:
Hi I'm still facing the same issue. I tried using SVO without ROS it worked fine but I'm not able to get this working with ROS. Any suggestions would be very helpful.

OS: Ubuntu 14.04
rosversion:  Indigo 1.11.21

ChaoWang-SCUT, 2017-05-05:
Hi, I'm also facing this issue, and my working environments is the same as you. I have been in this problem for a long time. Have you resolved the problem? Urgent need your help.

saikiran321, 2017-05-05:
No, I couldn't get it work with ROS. I'm using svo with c++

Zengqingfeng123, 2021-06-01:
I guess you should build the SVO file first.(If you starting build it, you will find more problems to deal with.)