uzh-rpg/rpg_svo #198: The relative distance and rotation between camera and IMU within the provided dataset

Booooooosh, 2017-05-04:
I am currently trying to do a sensor fusion between the IMU and the output of the svo, in order to get a more accurate estimation of the pose of the UAV. There is a ros package called ethzasl_sensor_fusion (http://wiki.ros.org/ethzasl_sensor_fusion) that performs ekf sensor fusion but it requires the relative distance and position of the camera. Since I am using the dataset (rpg.ifi.uzh.ch/datasets/airground_rig_s3_2013-03-18_21-38-48.bag), I can not find such information, either on website or the reference paper of the dataset.

Many thanks!

highlightz, 2017-05-05:
we acquire imu and image data using iphone cell phone, and calibrate the imu device and rgb camera on iphone using https://github.com/ethz-asl/kalibr.

Booooooosh, 2017-05-05:
So you are retrieving data directly from the camera and the IMU on the iphone, and using this ros package to calibrate the spacial parameters between the two devices, is that correct? @highlightz