uzh-rpg/rpg_svo #202: Segmentation fault (core dumped) occurs when run test_pipeline

chenchr, 2017-05-16:
a few days before i run the test_pipeline correctly, today, i pull the latest version of svo, compile and run test_pipe, it  show that : 
[INFO] SVO initialized
reading image /home/chenchr/Datasets/sin2_tex2_h1_v8_d/img/frame_000002_0.png
[INFO] RESET
[INFO] Init: Selected first frame.
Frame-Id: 0 	#Features: 0 	Proc. Time: 6.229ms 
[INFO] Init: KLT tracked 565 features
[INFO] Init: KLT 25.3823px average disparity.
Frame-Id: 1 	#Features: 0 	Proc. Time: 10.192ms 
[INFO] Init: KLT tracked 543 features
[INFO] Init: KLT 50.8035px average disparity.
Segmentation fault (core dumped)

then i use cout to locate where the error happens, when invokes processSecondFrame, it invokes KltHomographyInit::addSecondFrame, and then it will invoke
computeHomography, it set up the cout(means that i cout some text to remind me the program works) before that, which go fine. then i set up a cout at the
end of function computeHomography, it works too. however, i set up the cout in just following the function computeHomography which is called in KltHomographyInit::addSecondFrame,
the error "Segmentation fault (core dumped)" occurs again, i think the reason of error maybe have something to do with the function arguments. however i can't fix it. can anyone help me out ? 
thanks!

chenchr, 2017-05-16:
i recompile vikit and then svo, it works.

closed 2017-05-16 by chenchr

Zcyyy, 2019-03-14:
I also meet same problem.But no matter how i recompile vikit and svo,it did not work.

Beniko95J, 2019-04-06:
I also meet this problem. Due to the bt output, I think there may be some problems with this function vk::Homography::decompose(). Anyone has ideas?

(gdb) bt
#0  0x00007ffff674598d in __GI___libc_free (mem=0x51) at malloc.c:3103
#1  0x00007ffff7068607 in vk::Homography::decompose() ()
    at /home/benik/Documents/git/build/rpg_vikit/vikit_common/lib/libvikit_common.so
#2  0x00007ffff70695e6 in vk::Homography::computeSE3fromMatches() ()
    at /home/benik/Documents/git/build/rpg_vikit/vikit_common/lib/libvikit_common.so
#3  0x00007ffff7b839d7 in svo::initialization::computeHomography(std::vector<Eigen::Matrix<double, 3, 1, 0, 3, 1>, std::allocator<Eigen::Matrix<double, 3, 1, 0, 3, 1> > > const&, std::vector<Eigen::Matrix<double, 3, 1, 0, 3, 1>, std::allocator<Eigen::Matrix<double, 3, 1, 0, 3, 1> > > const&, double, double, std::vector<int, std::allocator<int> >&, std::vector<Eigen::Matrix<double, 3, 1, 0, 3, 1>, std::allocator<Eigen::Matrix<double, 3, 1, 0, 3, 1> > >&, Sophus::SE3&) () at /home/benik/Documents/git/build/rpg_svo/svo/lib/libsvo.so
#4  0x00007ffff7b851d0 in svo::initialization::KltHomographyInit::addSecondFrame(boost::shared_ptr<svo::Frame>) ()
    at /home/benik/Documents/git/build/rpg_svo/svo/lib/libsvo.so
#5  0x00007ffff7b56368 in svo::FrameHandlerMono::processSecondFrame() ()
    at /home/benik/Documents/git/build/rpg_svo/svo/lib/libsvo.so
#6  0x00007ffff7b57c21 in svo::FrameHandlerMono::addImage(cv::Mat const&, double) () at /home/benik/Documents/git/build/rpg_svo/svo/lib/libsvo.so
#7  0x00005555555615a9 in svo::BenchmarkNode::runFromFolder() ()
#8  0x0000555555560a79 in main ()

Beniko95J, 2019-04-06:
After I recompile vikit without -march=nactive, it works Haha.

WANGLXUE, 2019-06-14:
Save problem. Again and again. As @Beniko95J  suggested, problem changes. I pasted last three rows of problem. After move -march=nactive, problem changes from (1) to (2).
(1)
[INFO] Init: Selected second frame, triangulated initial map.
Frame-Id: 2 	#Features: 498 	Proc. Time: 8.258ms 
Segmentation fault (core dumped)
(2)
 [INFO] Init: KLT 50.8035px average disparity.
double free or corruption (out)
Aborted (core dumped)

xixi0226, 2019-08-01:
delete the sentence about "-march " in CMakeLists.txt

yizhang-group, 2023-09-14:
yes, as @xixi0226 and @Beniko95J said, delete "-march" in svo and vikit, and then recompile them.