uzh-rpg/rpg_svo #207: Low FPS on Raspberry

DxsGeorge, 2017-07-07:
I am running SVO on ROS, with live camera. On 1280x720 resolution I get about 6-10 FPS when I have enough features, then on fast movements features are not relocated and it gets lost. If I lower the resolution I get more FPS but lower features so I get lost easily again. Are there any settings I can mess with so that I get better perfomrance?

AlexisTM, 2017-09-05:
6-10 fps for vision odometry on a Pi3 is already a good score! 

I would change the board (Odroid XU4 should/could go up to 60FPS) or Intel Compute board (Intel aero).

AlexisTM, 2017-09-05:
Also, do you run the bundle adjustment? If you do, you can try to disable it to get more fps.

AlexisTM, 2017-09-06:
@DxsGeorge You can as well disable TRACE in the CMakeLists.txt of the `svo` folder and compile with the following:

```bash
rm -rf devel/ build/
export ARM_ARCHITECTURE=True
catkin_make -DCMAKE_BUILD_TYPE=Release
```

DxsGeorge, 2017-09-11:
@AlexisTM thanks for the tips! I am currently running another vo algorithm, if I get down to svo again, I will report any differences with these changes!

zanazakaryaie, 2017-09-18:
@DxsGeorge 
I compiled and tested it in Rpi2 which gave me 20FPS on its own dataset (sin2_tex2_h1_v8_d.tar.gz).
The reason is for enabling ARM_ARCHITECTURE and lower image resolution of the dataset. The bundle adjustment was also disabled.

The main problem of SVO is for forward motions. What package do you use now?

AlexisTM, 2017-09-18:
@zanazakaryaie I am investigating Rovio and SVO 2.0.

zanazakaryaie, 2017-09-18:
@AlexisTM Could you please explain about performance of SVO 2.0. Does it support forward motions?
What is your platform and how many FPS you got?

AlexisTM, 2017-09-18:
SVO 2.0 add the IMU input and any number of cameras. 

Advertised at 100fps on Odroid XU4, so maybe 30 fps on RPi3. Note: The ARMv7h binaries should arrive in a month. 

http://rpg.ifi.uzh.ch/svo2.html

I prefer ROVIO as it is free.

zanazakaryaie, 2017-09-18:
@AlexisTM Thanks for explanations. What OS you use on Odroid XU4? The SVO 2 binaries are only available for ubuntu 14.04 and 16.04.

One more question. Does ROVIO requires inter sensor calibration? How about synchronization?

AlexisTM, 2017-09-18:
SVO2 binaries are for intel x64 for now. 

On the Odroid XU4 I am running Ubuntu server 16.04.

AlexisTM, 2017-09-18:
ROVIO needs IMU/Camera calibration, done via Kalibr.

zanazakaryaie, 2017-09-18:
So many thanks for your explanations.

Just a matter of confusion. If binaries are for intel x64, then how could you run it on Odroid XU4? I couldn't understand what you mean by server.

DxsGeorge, 2017-09-19:
@zanazakaryaie I am using DSO with ROS now.

mdne, 2017-09-22:
has someone tried ROVIO on Raspberry Pi3 ?

AlexisTM, 2017-09-22:
I did not, yet, I'm not optimistic about the performances. When I checked it last year, it was much more computationally expensive. Yet, it integrates the IMU directly so it is more robust (if your camera is not blurry of course)

0xfaded, 2017-10-25:
Im currently working with orb slam so I don't know where svo's bottlenecks lie, but if you're interested my FAST detector is tuned for the raspberry pi 3 and will extract features from your 1240x720 image in ~5ms.

https://github.com/0xfaded/pislam

loseyourself-gan, 2021-03-04:
@AlexisTM
Hello, I encountered such a problem when compiling svo2.0 on raspberry pie. Do you know how to solve it？
error: ‘plus’ is not a member of ‘pcl::traits’
  566 | POINT_CLOUD_REGISTER_POINT_STRUCT (pcl::PPFSignature,

AlexisTM, 2021-03-04:
Sadly, I didn't use it for 2y. You should seek official documentation and support.

eremteknoloji, 2021-07-25:
Hi Alexis, What camera did you use with xu4 at 60fps?

AlexisTM, 2021-07-25:
This was some uEye global shutter camera, over 3 years ago ;)