uzh-rpg/rpg_svo #208: interrupted by signal 11: SIGSEGV (without ROS)

ahuo002, 2017-07-17:
[INFO] SVO initialized
---
reading image /home/github/SLAM/rpg_svo/svo/test/Datasets/sin2_tex2_h1_v8_d/img/frame_000002_0.png
[INFO] RESET
[INFO] New Frame
[INFO] Init: Selected first frame.
[INFO] Frame: 0	 fps-avg = inf	 nObs = 0
Frame-Id: 0 	#Features: 0 	Proc. Time: 6.103ms 
---
[INFO] New Frame
[INFO] Init: KLT tracked 565 features
Frame-Id: 1 	#Features: 0 	Proc. Time: 11.213ms 
---
[INFO] Init: KLT 25.3823px average disparity.
[INFO] Frame: 1	 fps-avg = 163.854	 nObs = 0
[INFO] New Frame
[INFO] Init: KLT tracked 543 features
[INFO] Init: KLT 50.8035px average disparity.
[INFO] Init: Homography RANSAC 543 inliers.
[INFO] Init: Selected second frame, triangulated initial map.
[INFO] Frame: 2	 fps-avg = 115.5	 nObs = 0
Frame-Id: 2 	#Features: 498 	Proc. Time: 9.184ms 

Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)

closed 2017-07-18 by ahuo002

jingwuOUO, 2019-05-25:
could you please tell me the solution to this problem?

DaddyWesker, 2021-08-23:
So, how did you fixed that?