uzh-rpg/rpg_svo #209: Optimizing SVO for Forward Facing Movements

atomoclast, 2017-07-27:
Hello,

On this page: https://github.com/uzh-rpg/rpg_svo/wiki/Obtaining-Best-Performance, you mention that SVO is not optimized for forward facing motions right now. I wanted to ask what exactly doesn't make it optimized for forward facing applications? If I wanted to optimize it for a forward facing application, what should be changed exactly to make it better for a forward facing application? I'm going to guess it's not a simple set of changes.

Thanks in advance! Great software otherwise!

YoshuaNava, 2017-07-28:
Hey,
Thanks for asking. I'm very interested in this too.

zanazakaryaie, 2017-09-18:
Hi

That's my question too. I tested SVO on kitti and Euroc datasets but it failed!! 

What should I do?

maazmb, 2018-01-02:
why isnt anyone replying to this..:(

zanazakaryaie, 2018-01-02:
Hello

They have solved the problem in SVO 2.0. 

It can support forward facing too. They have released binaries and no source code is provided. It is available under ROS

maazmb, 2018-01-17:
Hi,
thanks zanazakaryaie for your reply. Actually i am interested (for research purposes) in the source code of SVO 2.0 for forward facing movements but unfortunately that is not available yet. Has anyone tried working on making changes to the SVO algorithm for it to support forward facing camera?
Any insight would be helpful..:)

YoshuaNava, 2018-02-15:
Hi,
I gather that forward-motion support was implemented in PL-SVO, by Rubén Gómez Ojeda:

https://github.com/rubengooj/pl-svo