uzh-rpg/rpg_svo #210: [svo-1] process has died [pid 11179, exit code -11, cmd /home/mlab/catkin_ws/devel/lib/svo_ros/vo __name:=svo __log:=/home/mlab/.ros/log/7156adfe-8658-11e7-bfcf-1c6f6559cbe5/svo-1.log].

ppls01, 2017-08-21:
Hi
I use Ubuntu 14.04 32bits and ROS indigo.
I try to run test_rig3.launch with example data given by their package.
But process stops and shuts down some error.

this my output terminal:
----------------------------------------
SUMMARY
========

CLEAR PARAMETERS
 * /svo/

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /svo/cam_cx: 0.45905
 * /svo/cam_cy: 0.510056
 * /svo/cam_d0: 0.932
 * /svo/cam_fx: 0.509326
 * /svo/cam_fy: 0.796651
 * /svo/cam_height: 480
 * /svo/cam_model: ATAN
 * /svo/cam_topic: /camera/image_raw
 * /svo/cam_width: 752
 * /svo/grid_size: 30
 * /svo/init_rx: 3.14
 * /svo/init_ry: 0.0
 * /svo/init_rz: 0.0
 * /svo/loba_num_iter: 0
 * /svo/max_n_kfs: 10

NODES
  /
    svo (svo_ros/vo)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[svo-1]: started with pid [12200]
create vo_node
[ WARN] [1503311715.578530749]: Cannot find value for parameter: svo/publish_img_pyr_level, assigning default: 0
[ WARN] [1503311715.580467658]: Cannot find value for parameter: svo/publish_every_nth_img, assigning default: 1
[ WARN] [1503311715.583482082]: Cannot find value for parameter: svo/publish_every_nth_dense_input, assigning default: 1
[ WARN] [1503311715.586479205]: Cannot find value for parameter: svo/publish_world_in_cam_frame, assigning default: 1
[ WARN] [1503311715.587775003]: Cannot find value for parameter: svo/publish_map_every_frame, assigning default: 0
[ WARN] [1503311715.589271790]: Cannot find value for parameter: svo/publish_point_display_time, assigning default: 0
[ WARN] [1503311715.953991689]: Cannot find value for parameter: svo/publish_markers, assigning default: 1
[ WARN] [1503311715.955407648]: Cannot find value for parameter: svo/publish_dense_input, assigning default: 0
[ WARN] [1503311715.957529690]: Cannot find value for parameter: svo/accept_console_user_input, assigning default: 1
[ INFO] [1503311715.959459766]: Found parameter: svo/cam_model, value: ATAN
[ INFO] [1503311715.961143718]: Found parameter: svo/cam_width, value: 752
[ INFO] [1503311715.962937703]: Found parameter: svo/cam_height, value: 480
[ INFO] [1503311715.964293701]: Found parameter: svo/cam_fx, value: 0.509326
[ INFO] [1503311715.965476712]: Found parameter: svo/cam_fy, value: 0.796651
[ INFO] [1503311715.966746172]: Found parameter: svo/cam_cx, value: 0.45905
[ INFO] [1503311715.968046836]: Found parameter: svo/cam_cy, value: 0.510056
[ INFO] [1503311715.969237699]: Found parameter: svo/cam_d0, value: 0.932
[ WARN] [1503311715.970453053]: Cannot find value for parameter: svo/init_tz, assigning default: 0
[ WARN] [1503311715.971856996]: Cannot find value for parameter: svo/init_ty, assigning default: 0
[ WARN] [1503311715.975969804]: Cannot find value for parameter: svo/init_tx, assigning default: 0
[ INFO] [1503311715.977397516]: Found parameter: svo/init_rz, value: 0
[ INFO] [1503311715.978874989]: Found parameter: svo/init_ry, value: 0
[ INFO] [1503311715.980061970]: Found parameter: svo/init_rx, value: 3.14
[ WARN] [1503311715.982903449]: Cannot find value for parameter: svo/trace_name, assigning default: svo
[ WARN] [1503311715.984735695]: Cannot find value for parameter: svo/trace_dir, assigning default: /tmp
[ WARN] [1503311715.985873784]: Cannot find value for parameter: svo/n_pyr_levels, assigning default: 3
[ WARN] [1503311715.987218048]: Cannot find value for parameter: svo/use_imu, assigning default: 0
[ WARN] [1503311715.988565431]: Cannot find value for parameter: svo/core_n_kfs, assigning default: 3
[ WARN] [1503311715.989891996]: Cannot find value for parameter: svo/map_scale, assigning default: 1
[... 67 lines cut ...]
[ INFO] [1503311716.981712298]: Init: KLT tracked 166 features
[ INFO] [1503311716.981830428]: Init: KLT 40.5347px average disparity.
[ INFO] [1503311717.019158806]: Init: KLT tracked 165 features
[ INFO] [1503311717.019259919]: Init: KLT 40.4152px average disparity.
[ INFO] [1503311717.048778779]: Init: KLT tracked 164 features
[ INFO] [1503311717.048883959]: Init: KLT 39.7602px average disparity.
[ INFO] [1503311717.082488551]: Init: KLT tracked 162 features
[ INFO] [1503311717.082595154]: Init: KLT 41.3884px average disparity.
[ INFO] [1503311717.122840437]: Init: KLT tracked 158 features
[ INFO] [1503311717.122941925]: Init: KLT 44.3843px average disparity.
[ INFO] [1503311717.150564257]: Init: KLT tracked 157 features
[ INFO] [1503311717.150692118]: Init: KLT 47.7998px average disparity.
[ INFO] [1503311717.184628042]: Init: KLT tracked 156 features
[ INFO] [1503311717.184729337]: Init: KLT 51.1666px average disparity.
[ INFO] [1503311717.191076617]: Init: Homography RANSAC 144 inliers.
[svo-1] process has died [pid 12200, exit code -11, cmd /home/mlab/catkin_ws/devel/lib/svo_ros/vo __name:=svo __log:=/home/mlab/.ros/log/5f089668-865c-11e7-ad91-1c6f6559cbe5/svo-1.log].
log file: /home/mlab/.ros/log/5f089668-865c-11e7-ad91-1c6f6559cbe5/svo-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

-----------------------------------------------

Please check my error and give me answer.

enteguo, 2017-10-30:
so am i

Adu143, 2018-01-09:
Hello, I am also facing the same error. When Svo is run for few minutes, it crashes saying "process has died" with exit code -11. Any solutions??

djt11306, 2018-01-25:
I have similar problem, it crashes after running for a few seconds

djt11306, 2018-01-26:
@ppls01 I guess this problem is caused by the conflict of using 32 bit OS

CrisGao, 2019-12-21:
same problem.I can't understand why running datasets will report errors

faustoTapia, 2020-06-26:
Same problem here. Tried logging but it does not give anything useful. Does anybody have any updates?

Ciao-li, 2021-07-21:
I have encountered this problem before, and the reason for this problem is the wrong version of eigen. My original version is 3.3.4, and it can work normally after changing to 3.2.8. By the way, before changing the version, you'd better uninstall the original version. If this method does not solve your problem, it may be due to other reasons.