uzh-rpg/rpg_svo #212: [svo-2] process has died

wowdeath, 2017-10-14:
when I "roslaunch svo_ros live.launch"  
liu@liu-pc:~$ roslaunch svo_ros live.launch 
... logging to /home/liu/.ros/log/d8edaf76-b08d-11e7-b3f5-642737f495b9/roslaunch-liu-pc-14156.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:43576/

SUMMARY
========

CLEAR PARAMETERS
 * /svo/

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /svo/cam_cx: 320.5
 * /svo/cam_cy: 240.5
 * /svo/cam_d0: -0.283076
 * /svo/cam_d1: 0.066674
 * /svo/cam_d2: 0.000896
 * /svo/cam_d3: 0.000778
 * /svo/cam_fx: 268.51167803
 * /svo/cam_fy: 268.51167803
 * /svo/cam_height: 480
 * /svo/cam_model: Pinhole
 * /svo/cam_topic: /downward_cam/cam...
 * /svo/cam_width: 640
 * /svo/grid_size: 30
 * /svo/init_rx: 3.14
 * /svo/init_ry: 0.0
 * /svo/init_rz: 0.0
 * /svo/kfselect_mindist: 0.1
 * /svo/loba_num_iter: 0
 * /svo/max_fts: 200
 * /svo/max_n_kfs: 15
 * /svo/quality_max_drop_fts: 60

NODES
  /
    odometry_rviz (rviz/rviz)
    svo (svo_ros/vo)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[odometry_rviz-1]: started with pid [14178]
process[svo-2]: started with pid [14180]
create vo_node
[ WARN] [1507950907.679400700]: Cannot find value for parameter: svo/publish_img_pyr_level, assigning default: 0
[ WARN] [1507950907.682486858]: Cannot find value for parameter: svo/publish_every_nth_img, assigning default: 1
[ WARN] [1507950907.685337404]: Cannot find value for parameter: svo/publish_every_nth_dense_input, assigning default: 1
[ WARN] [1507950907.691473087]: Cannot find value for parameter: svo/publish_world_in_cam_frame, assigning default: 1
[ WARN] [1507950907.693243151]: Cannot find value for parameter: svo/publish_map_every_frame, assigning default: 0
[ WARN] [1507950907.695186900]: Cannot find value for parameter: svo/publish_point_display_time, assigning default: 0
[ WARN] [1507950908.571535038, 59.494000000]: Cannot find value for parameter: svo/publish_markers, assigning default: 1
[ WARN] [1507950908.573777402, 59.494000000]: Cannot find value for parameter: svo/publish_dense_input, assigning default: 0
[ WARN] [1507950908.577563283, 59.497000000]: Cannot find value for parameter: svo/accept_console_user_input, assigning default: 1
[ INFO] [1507950908.580245295, 59.497000000]: Found parameter: svo/cam_model, value: Pinhole
[ INFO] [1507950908.606421199, 59.509000000]: Found parameter: svo/cam_width, value: 640
[ INFO] [1507950908.616868625, 59.517000000]: Found parameter: svo/cam_height, value: 480
[ INFO] [1507950908.638242567, 59.528000000]: Found parameter: svo/cam_fx, value: 268.512
[ INFO] [1507950908.644204707, 59.528000000]: Found parameter: svo/cam_fy, value: 268.512
[ INFO] [1507950908.654664105, 59.532000000]: Found parameter: svo/cam_cx, value: 320.5
[ INFO] [1507950908.668981708, 59.541000000]: Found parameter: svo/cam_cy, value: 240.5
[ INFO] [1507950908.680187156, 59.547000000]: Found parameter: svo/cam_d0, value: -0.283076
[ INFO] [1507950908.693257895, 59.548000000]: Found parameter: svo/cam_d1, value: 0.066674
[ INFO] [1507950908.697191830, 59.549000000]: Found parameter: svo/cam_d2, value: 0.000896
[ INFO] [1507950908.699968923, 59.549000000]: Found parameter: svo/cam_d3, value: 0.000778
[ WARN] [1507950908.755755581, 59.568000000]: Cannot find value for parameter: svo/init_tz, assigning default: 0
[ WARN] [1507950908.760319540, 59.572000000]: Cannot find value for parameter: svo/init_ty, assigning default: 0
[ WARN] [1507950908.780632030, 59.580000000]: Cannot find value for parameter: svo/init_tx, assigning default: 0
[ INFO] [1507950908.782648664, 59.581000000]: Found parameter: svo/init_rz, value: 0
[ INFO] [1507950908.785686946, 59.582000000]: Found parameter: svo/init_ry, value: 0
[ INFO] [1507950908.788206603, 59.584000000]: Found parameter: svo/init_rx, value: 3.14
[ WARN] [1507950908.791059039, 59.584000000]: Cannot find value for parameter: svo/trace_name, assigning default: svo
[ WARN] [1507950908.793761839, 59.584000000]: Cannot find value for parameter: svo/trace_dir, assigning default: /tmp
[ WARN] [1507950908.796399869, 59.586000000]: Cannot find value for parameter: svo/n_pyr_levels, assigning default: 3
[ WARN] [1507950908.798477336, 59.587000000]: Cannot find value for parameter: svo/use_imu, assigning default: 0
[ WARN] [1507950908.800525119, 59.587000000]: Cannot find value for parameter: svo/core_n_kfs, assigning default: 3
[ WARN] [1507950908.802519262, 59.587000000]: Cannot find value for parameter: svo/map_scale, assigning default: 1
[ INFO] [1507950908.806098586, 59.588000000]: Found parameter: svo/grid_size, value: 30
[ WARN] [1507950908.809387958, 59.590000000]: Cannot find value for parameter: svo/init_min_disparity, assigning default: 50
[ WARN] [1507950908.812440866, 59.592000000]: Cannot find value for parameter: svo/init_min_tracked, assigning default: 50
[ WARN] [1507950908.815330598, 59.593000000]: Cannot find value for parameter: svo/init_min_inliers, assigning default: 40
[ WARN] [1507950908.818173707, 59.593000000]: Cannot find value for parameter: svo/klt_max_level, assigning default: 4
[ WARN] [1507950908.821896725, 59.593000000]: Cannot find value for parameter: svo/klt_min_level, assigning default: 2
[ WARN] [1507950908.830230388, 59.597000000]: Cannot find value for parameter: svo/reproj_thresh, assigning default: 2
[ WARN] [1507950908.837254371, 59.602000000]: Cannot find value for parameter: svo/poseoptim_thresh, assigning default: 2
[ WARN] [1507950908.853808184, 59.607000000]: Cannot find value for parameter: svo/poseoptim_num_iter, assigning default: 10
[ WARN] [1507950908.866247870, 59.612000000]: Cannot find value for parameter: svo/structureoptim_max_pts, assigning default: 20
[ WARN] [1507950908.886248745, 59.622000000]: Cannot find value for parameter: svo/structureoptim_num_iter, assigning default: 5
[ WARN] [1507950908.894092569, 59.630000000]: Cannot find value for parameter: svo/loba_thresh, assigning default: 2
[ WARN] [1507950908.896904872, 59.632000000]: Cannot find value for parameter: svo/loba_robust_huber_width, assigning default: 1
[ INFO] [1507950908.900222947, 59.635000000]: Found parameter: svo/loba_num_iter, value: 0
[ INFO] [1507950908.905887144, 59.638000000]: Found parameter: svo/kfselect_mindist, value: 0.1
[ WARN] [1507950908.909217643, 59.639000000]: Cannot find value for parameter: svo/triang_min_corner_score, assigning default: 20
[ WARN] [1507950908.927880774, 59.649000000]: Cannot find value for parameter: svo/triang_half_patch_size, assigning default: 4
[ WARN] [1507950908.934066600, 59.649000000]: Cannot find value for parameter: svo/subpix_n_iter, assigning default: 10
[ INFO] [1507950908.937576520, 59.650000000]: Found parameter: svo/max_n_kfs, value: 15
[ WARN] [1507950908.944139333, 59.654000000]: Cannot find value for parameter: svo/img_imu_delay, assigning default: 0
[ INFO] [1507950908.947668867, 59.654000000]: Found parameter: svo/max_fts, value: 200
[ WARN] [1507950908.949374140, 59.654000000]: Cannot find value for parameter: svo/quality_min_fts, assigning default: 50
[ INFO] [1507950908.951889535, 59.654000000]: Found parameter: svo/quality_max_drop_fts, value: 60
[ INFO] [1507950908.952324209, 59.654000000]: SVO initialized
[ INFO] [1507950908.970817574, 59.672000000]: Found parameter: svo/cam_topic, value: /downward_cam/camera/image
[ INFO] [1507950909.334644420, 59.917000000]: RESET
[svo-2] process has died [pid 14180, exit code -4, cmd /home/liu/svo_ws/devel/lib/svo_ros/vo __name:=svo __log:=/home/liu/.ros/log/d8edaf76-b08d-11e7-b3f5-642737f495b9/svo-2.log].
log file: /home/liu/.ros/log/d8edaf76-b08d-11e7-b3f5-642737f495b9/svo-2*.log

enteguo, 2017-11-02:
I fix it. carsh with opencv, make uninstall opencv

zainmehdi, 2017-11-02:
@enteguo can you please describe in detail how you fixed it ? Did you uninstall ros version of opencv ?

enteguo, 2017-11-02:
@zainmehdi ros version is indigo, opencv 3.2.0. only make uninstall opencv, then the error is fixed