uzh-rpg/rpg_svo #214: ld returned 1 exit status

femouse, 2017-11-30:
[ 66%] Built target svo
Linking CXX executable ../bin/test_depth_filter
/usr/bin/ld: CMakeFiles/test_depth_filter.dir/test/test_depth_filter.cpp.o: undefined reference to symbol '_ZN6Sophus3SE3aSERKS0_'
/usr/local/lib/libSophus.so: error adding symbols: DSO missing from command line
collect2: error: **# ld returned 1 exit status**
make[2]: *** [../bin/test_depth_filter] 错误 1
make[1]: *** [CMakeFiles/test_depth_filter.dir/all] 错误 2
make: *** [all] 错误 2
leo@leo:~/projectFile/ImagecodeProject/rpg_svo/svo/build$ make
[ 66%] Built target svo

femouse, 2017-11-30:
http://blog.csdn.net/shangzh/article/details/49079685
DSO missing from command line原因及解决办法
当binutils>=2.22时，编译出错了，ld会报上面的错.需要显示地链接库才可以通过。

# Create svo library
set(Sophus_LIBRARIES libSophus.so)
ADD_LIBRARY(svo SHARED ${SOURCEFILES})
TARGET_LINK_LIBRARIES(svo 
${Sophus_LIBRARIES}
${LINK_LIBS}

)

genghe1991, 2018-03-21:
I came across the same problem. I have solved it with the help of this comment.
The problem seems to be caused by libSophus.so. 
The comment above does provide a right direction to solve problem. But it lacks some accuracy and a few details on the solution.  And the Chinese characters may also confuse a lot people who can not read Chinese. 
Therefore I would like to post more to help people who encounter the DSO missing problem.

As shown in the error, the problem is related with libSophus.so. It seems it could not link to the Sophus lib correctly. (BTW, I have encountered another error also related to Sophus when building vikit_common in this project. It is also caused by the link stuff and the solution is actually more or less the same)

To solve the problem, just edit the CMakeList.txt under the rpg_svo/svo directory.
Go to line 79(in my case) and find  # Set link libraries
After that ,add 

set(Sophus_LIBRARIES libSophus.so)

before 
LIST(APPEND LINK_LIBS 
  ${OpenCV_LIBS}
  ${Sophus_LIBRARIES}
  ${Boost_LIBRARIES}
  ${fast_LIBRARIES}
)

All good after that for me. :)