uzh-rpg/rpg_svo #216: global or rolling shutter

kamel271, 2017-12-19:
hello guys i'm trying to implement svo 2.0 in an robot with v2 camera module of raspberry pi 3   the thing is that  this   camera is with a rolling shutter  so is that possible ? because  in the guide they said be sure of using a global shutter camera

zanazakaryaie, 2018-01-16:
I have tested it with v1 camera module which is also rolling shutter. It works but if you want to get the best performance you should use global shutter.

Increasing frame rate of your camera may help to reduce rolling shutter effect

How do you want to resolve scale?

deepak-1530, 2019-03-27:
How was the performance using picam?

kamel271, 2019-03-27:
It is quite nice you can use kinect also

deepak-1530, 2019-03-28:
I am currently applying it on a drone using picam. Can you send your code
that you used on raspi. It will be really helpful.

Regards
Deepak Singh