uzh-rpg/rpg_svo #223: catkin_make error

cumtchenchang, 2018-03-12:
home/cc/tem_ws/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::exp(Eigen::Matrix<double, 6, 1, 0, 6, 1> const&)'
/home/cc/tem_ws/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::operator=(Sophus::SE3 const&)'
/home/cc/tem_ws/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::SE3()'
/home/cc/tem_ws/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::SE3(Sophus::SE3 const&)'
/home/cc/tem_ws/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::operator*(Sophus::SE3 const&) const'
/home/cc/tem_ws/devel/lib/libvikit_common.so: undefined reference to `Sophus::SO3::matrix() const'
/home/cc/tem_ws/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::SE3(Eigen::Matrix<double, 3, 3, 0, 3, 3> const&, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)'
/home/cc/tem_ws/devel/lib/libvikit_common.so: undefined reference to `Sophus::SE3::operator*(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&) const'
collect2: error: ld returned 1 exit status
rpg_vikit/vikit_common/CMakeFiles/test_vk_common_triangulation.dir/build.make:178: recipe for target '/home/cc/tem_ws/devel/lib/vikit_common/test_vk_common_triangulation' failed
make[2]: *** [/home/cc/tem_ws/devel/lib/vikit_common/test_vk_common_triangulation] Error 1
CMakeFiles/Makefile2:1491: recipe for target 'rpg_vikit/vikit_common/CMakeFiles/test_vk_common_triangulation.dir/all' failed
make[1]: *** [rpg_vikit/vikit_common/CMakeFiles/test_vk_common_triangulation.dir/all] Error 2
[ 56%] Building CXX object rpg_svo/svo/CMakeFiles/svo.dir/src/point.cpp.o
[ 57%] Building CXX object rpg_svo/svo/CMakeFiles/svo.dir/src/map.cpp.o
[ 59%] Building CXX object rpg_svo/svo/CMakeFiles/svo.dir/src/pose_optimizer.cpp.o
[ 60%] Building CXX object rpg_svo/svo/CMakeFiles/svo.dir/src/initialization.cpp.o
[ 61%] Building CXX object rpg_svo/svo/CMakeFiles/svo.dir/src/matcher.cpp.o
[ 63%] Building CXX object rpg_svo/svo/CMakeFiles/svo.dir/src/reprojector.cpp.o
[ 64%] Building CXX object rpg_svo/svo/CMakeFiles/svo.dir/src/feature_alignment.cpp.o
[ 65%] Building CXX object rpg_svo/svo/CMakeFiles/svo.dir/src/feature_detection.cpp.o
[ 67%] Building CXX object rpg_svo/svo/CMakeFiles/svo.dir/src/depth_filter.cpp.o
[ 68%] Building CXX object rpg_svo/svo/CMakeFiles/svo.dir/src/config.cpp.o
[ 69%] Building CXX object rpg_svo/svo/CMakeFiles/svo.dir/src/sparse_img_align.cpp.o
[ 71%] Linking CXX shared library /home/cc/tem_ws/devel/lib/libsvo.so
[ 71%] Built target svo
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

Flock1, 2018-05-03:
Try this:
```
cd ~/simulation/ros_catkin_ws    
catkin_make --pkg mav_msgs
source devel/setup.bash
```

LIANGXINKAI, 2018-05-29:
add "SET(Sophus_LIBRARIES libSophus.so " to /catkin_ws/src/rgp_vikit/vikit_common/CMakeLists.txt
such as
[image]

smorad, 2019-06-03:
When you build sophus, do `make install` at the end. It should print a line `-- Installing: /usr/local/lib/libSophus.so ` towards the end.

Go to the line @LIANGXINKAI references and add that path to the `TARGET_LINK_LIBRARIES` command:

```
 ADD_LIBRARY(${PROJECT_NAME} SHARED ${SOURCEFILES})
 TARGET_LINK_LIBRARIES(${PROJECT_NAME}
   ${OpenCV_LIBS}
   ${Sophus_LIBRARIES}
   ${catkin_LIBRARIES}
   /usr/local/lib/libSophus.so
) 
```

I also needed to add the path in `rpg_svo/svo/CMakeLists.txt`

`TARGET_LINK_LIBRARIES(svo ${LINK_LIBS} /usr/local/lib/libSophus.so)`

cumtchenchang, 2019-10-14:
@smorad @LIANGXINKAI Thank you for your answer, I am sorry for the late reply.  Today, I install svo and encounter the same error. I followed your suggestion, but the error is still on. Thank you.

cumtchenchang, 2019-10-14:
I have solved. add SET(Sophus_LIBRARIES libSophus.so) in the front of CMakeLists.txt. Thank you.

closed 2019-10-14 by cumtchenchang