uzh-rpg/rpg_svo #229: Tested code with live camera

weidongguo, 2018-04-14:
Our team has tested the code (ROS version) with a live camera. We coudn't reproduce the same performance as if the ROS bag file was replayed. 

Here is the result:
https://drive.google.com/file/d/19Qy4BdTpAHXQZo3rexffL6604cIkSkC_/view?usp=sharing

We notice that if we move the camera fast, most of the features points are lost, and therefore the pose estimation is no longer accurate. To help it, we can make the camera still and wait until it gets enough features points again. At that point in time, if we make a movement, it will be smooth. Immediately after we have gone far enough that the features points aren't enough anymore (basically, current frame so different from the key frame), the pose estimation is again no longer accurate.

On other other hand, if we simply replay the bag file as provided by the source code, the pose estimation is so accurate even when the camera moves fast.

Are there certain settings that we omit?

canberkgurel, 2019-06-21:
Hi @weidongguo @odinhr 

I have the exact same problem. Did you manage to solve this issue?

skandaswaroop1998, 2019-12-05:
Hi @weidongguo @odinhr I am facing the same problem. It is not recognizing key features properly from my camera even though it is quite a feature rich environment that I am testing in. Any ideas on how to improve it?

EduardoSH94, 2020-03-26:
Any solution? I have the same issue.