uzh-rpg/rpg_svo #232: Is SVO compatible with Ubuntu 16.04?

Flock1, 2018-05-01:


Flock1, 2018-05-02:
@odinhr,
Thanks for the reply. I'm using Lunar. So I guess that's the problem.

Flock1, 2018-05-03:
@odinhr, This is what I'm getting when I run the .bag file (final step):

```
[svo-1] process has died [pid 22806, exit code -11, cmd /home/sarvagya/catkin_ws/devel/lib/svo_ros/vo __name:=svo __log:=/home/sarvagya/.ros/log/740d9b28-4ec8-11e8-be1c-7085c259af94/svo-1.log].
log file: /home/sarvagya/.ros/log/740d9b28-4ec8-11e8-be1c-7085c259af94/svo-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete

```

odinhr, 2018-05-04:
Maybe you should try with this commit. It could be better for you. Sometimes I have similars problems before that commit.

https://github.com/uzh-rpg/rpg_svo/commit/ec240827b1b284fb3269e2b39d5871d92fd5d19f

odinhr, 2018-05-11:
I have similar problems running SVO on Raspberry Pi. I solved the problems running make on catkin_ws/build. I compiled SVO on pc long time ago. I don't remember my problems. @Flock1

saikrn112, 2018-05-13:
@Flock1 
Is this issue solved? I am getting similar problems as you.

Flock1, 2018-05-13:
@saikrn112, I couldn't. I'm getting this make file error as shown above.

mhubii, 2018-09-04:
Well I had a similar problem and as supposed in #174, installing Eigen 3.2.10 fixed it for me. If you don't need your current Eigen version, then simply follow #174. But you can also do a local installation, then follow these steps:

Clone Eigen (e.g. to your Downloads folder) from https://bitbucket.org/eigen/eigen/src/default/
Then checkout branch 3.2.10

´cd ~/Downloads/eigen && hg checkout 3.2.10´

To install Eigen locally, build it like follows:

´cd ~/Downloads/eigen && mkdir build && cmake -DCMAKE_INSTALL_PREFIX=${HOME}/local .. && make install´

Now re-build SVO.
From your catkin_ws folder remove the build folder, the devel folder, and src/CMakeLists.txt

´cd <path to your catkin_ws> && rm -r build && rm -r devel && rm src/CMakeLists.txt´

Now, since you want to use your locally installed Eigen, you need to build SVO by hand. 

´cd <path to your catkin_ws>/src && catkin_init_workspace && cd .. && mkdir build && cd build´

Then make it with your local Eigen install e.g.:

´cmake -DEIGEN_INCLUDE_DIR=${HOME}/local/include/eigen3 -DCMAKE_INSTALL_PREFIX=../install -DCATKIN_DEVEL_PREFIX=../devel ../src´

That should fix it.